adafruit/circuitpython #777: samd21: Is OSC32K disabled?

notro, 2018-04-25:
I'm trying to calibrate the rtc on the Circuit Playground Express, but since it's way off, I can't do it with the rtc calibration. So I'm looking at calibrating the clock: OSC32K -> GEN2 -> GCLK_RTC.

Looking at the (really convoluted asf4 configuration) code it seemed that OSC32K was disabled, which was confirmed by checking the OSC32K register value.
So I enabled it (which should also result in the NVM calibration value being loaded):
```
diff --git a/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h b/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h
-#define CONF_OSC32K_CONFIG 0
+#define CONF_OSC32K_CONFIG 1

-#define CONF_OSC32K_ENABLE 0
+#define CONF_OSC32K_ENABLE 1
```
This resulted in a bright green led and no usb devices showing up.

Does anyone know what's going on here? 
How can the rtc be running with the clock disabled?

Here's a register dump (adapted from [ZeroRegs.cpp](https://github.com/drewfish/arduino-ZeroRegs/blob/master/src/ZeroRegs.cpp)):
```
Adafruit CircuitPython 3.0.0-alpha.6-dirty on 2018-04-25; Adafruit CircuitPlayground Express with samd21g18

--------------------------- GCLK
GCLK_MAIN:  GEN00 (always)
GCLK_DFLL48M_REF:  --disabled--
GCLK_DPLL:  --disabled--
GCLK_DPLL_32K:  --disabled--
GCLK_WDT:  --disabled--
GCLK_RTC:  GEN02
GCLK_EIC:  --disabled--
GCLK_USB:  GEN00
GCLK_EVSYS_CHANNEL_0:  --disabled--
GCLK_EVSYS_CHANNEL_1:  --disabled--
GCLK_EVSYS_CHANNEL_2:  --disabled--
GCLK_EVSYS_CHANNEL_3:  --disabled--
GCLK_EVSYS_CHANNEL_4:  --disabled--
GCLK_EVSYS_CHANNEL_5:  --disabled--
GCLK_EVSYS_CHANNEL_6:  --disabled--
GCLK_EVSYS_CHANNEL_7:  --disabled--
GCLK_EVSYS_CHANNEL_8:  --disabled--
GCLK_EVSYS_CHANNEL_9:  --disabled--
GCLK_EVSYS_CHANNEL_10:  --disabled--
GCLK_EVSYS_CHANNEL_11:  --disabled--
GCLK_SERCOMx_SLOW:  --disabled--
GCLK_SERCOM0_CORE:  --disabled--
GCLK_SERCOM1_CORE:  --disabled--
GCLK_SERCOM2_CORE:  --disabled--
GCLK_SERCOM3_CORE:  GEN00
GCLK_SERCOM4_CORE:  --disabled--
GCLK_SERCOM5_CORE:  --disabled--
GCLK_TCC0_TCC1:  --disabled--
GCLK_TCC2_TC3:  --disabled--
GCLK_TC4_TC5:  --disabled--
GCLK_TC6_TC7:  --disabled--
GCLK_ADC:  --disabled--
GCLK_AC_DIG:  --disabled--
GCLK_AC_ANA:  --disabled--
GCLK_DAC:  --disabled--
GCLK_PTC:  --disabled--
GCLK_I2S_0:  --disabled--
GCLK_I2S_1:  --disabled--
GEN00:  DFLL48M OE
GEN01:  DFLL48M/150 OE
GEN02:  OSC32K/32
GEN03:  --disabled--
GEN04:  --disabled--
GEN05:  --disabled--
GEN06:  --disabled--
GEN07:  --disabled--
GEN08:  --disabled--
--------------------------- SYSCTRL
PCLKSR:  DFLLRDY DFLLLCKF DFLLLCKC BOD33RDY B33SRDY
XOSC:  --disabled--
XOSC32K:  --disabled--
OSC32K:  --disabled--
OSCULP32K:  CALIB=0x17
OSC8M:  ONDEMAND PRESC=8 CALIB=0x817 FRANGE=8-11MHz
DFLL:  MODE USBCRM CCDIS
BOD33:  ACTION=RESET PSEL=1 LEVEL=0x7
VREG:
VREF:  CALIB=0x60
DPLL:  --disabled--
--------------------------- RTC MODE0
CTRL:  PRESCALER=1024
READREQ:
EVCTRL:
FREQCORR:  0x80
COMP0:  0x0
```

The OSC32K is briefly enabled as this snippet from asf4/samd21/hpl/sysctrl/hpl_sysctrl.c shows:
```c
void _sysctrl_init_sources(void)
{
...
#if CONF_OSC32K_CONFIG == 1
	/* OSC32K calibration value at bit 44:38 of memory 0x00806020 */
	calib = (*((uint32_t *)0x00806024) & 0x0001FC0) >> 6;

	hri_sysctrl_write_OSC32K_reg(hw,
#if CONF_OSC32K_OVERWRITE_CALIBRATION == 1
	                             SYSCTRL_OSC32K_CALIB(CONF_OSC32K_CALIB) |
#else
	                             SYSCTRL_OSC32K_CALIB(calib) |
#endif
	                                 (CONF_OSC32K_WRTLOCK << SYSCTRL_OSC32K_WRTLOCK_Pos)
	                                 | SYSCTRL_OSC32K_STARTUP(CONF_OSC32K_STARTUP)
	                                 | (CONF_OSC32K_RUNSTDBY << SYSCTRL_OSC32K_RUNSTDBY_Pos)
	                                 | (CONF_OSC32K_EN1K << SYSCTRL_OSC32K_EN1K_Pos)
	                                 | (CONF_OSC32K_EN32K << SYSCTRL_OSC32K_EN32K_Pos)
	                                 | (1 << SYSCTRL_OSC32K_ENABLE_Pos));
#else
	/* Enable OSC32K anyway since GCLK configuration may need it to sync */
	hri_sysctrl_set_OSC32K_ENABLE_bit(hw);
#endif
...
}

void _sysctrl_init_referenced_generators(void)
{
...
#if CONF_OSC32K_CONFIG == 0 || CONF_OSC32K_ENABLE == 0
	/* Disable after all possible configurations needs sync written. */
	hri_sysctrl_clear_OSC32K_ENABLE_bit(hw);
#endif
...
}

```

tannewt, 2018-04-25:
Not sure how it'd be running with OSC32K disabled.

In the long term, I'd like to move away from ASF4 for clocks because it is confusing. I started this file that we could expand: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/clocks.c

dhalbert, 2018-04-25:
There is also `#define CONF_OSC32K_EN32K 0` in `hpl_sysctrl_config.h` which probably should be set to 1. OSC32K is set to be the source for GCLK2, which is fed to the RTC: `#define CONF_GCLK_RTC_SRC GCLK_CLKCTRL_GEN_GCLK2_Val`. However, in `hpl_gclk_config.h`, GCLK2 Output Enable is turned off: `#define CONF_GCLK_GEN_2_OE 0`.

I don't know why incorrect settings would make it hang. It would be worth looking at the stack backtrace in gdb when it doesn't seem to come up.

I myself also tried enabling OSC32K while also trying to get XOSC32K to work on boards that have a crystal, and ran into similar problems as you did. I am not impressed by the clock setup code in ASF4: there are dependencies between the clocks that may be an issue. Looking at the Arduino clock setup code may be instructive.

dhalbert, 2018-04-25:
We would also like to use OSC32K or XOSC32K (when available) as a reference for DFLL48M when the USB clock is not available. However, running DFLL48M in closed-loop mode is broken in SAMD51 (see the errata).

sommersoft, 2018-04-25:
ASF4 definitely contains zero fun.

Having little experience with the RTC and clocks, I only rolled back your changes, and used the [Ultra Low Power OSC32K](https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h#L380).

I would guess that leaving the OSCULP32K enable while turning on the OSC32K might cause some issues. 🤷‍

If the ULP can't be calibrated as well as the non-ULP, could definitely switch to that...

ladyada, 2018-04-25:
pinging @deanm1278 as well since he knows some of this clock nonsense

notro, 2018-04-25:
> There is also #define CONF_OSC32K_EN32K 0 in hpl_sysctrl_config.h which probably should be set to 1. 

Thanks, that did it.

Without any manual calibration I'm now at -8000 ppm, down from -100000 ppm
```
-0.023659 secs / 0.000053 days = -442.667056 secs/day (-5123.46 ppm) : 25 Apr 2018 21:34:01
-0.491396 secs / 0.000742 days = -662.498077 secs/day (-7667.80 ppm) : 25 Apr 2018 21:35:00
-0.974833 secs / 0.001442 days = -676.135505 secs/day (-7825.64 ppm) : 25 Apr 2018 21:36:00
-1.451387 secs / 0.002130 days = -681.352201 secs/day (-7886.02 ppm) : 25 Apr 2018 21:36:59
-1.945305 secs / 0.002830 days = -687.309458 secs/day (-7954.97 ppm) : 25 Apr 2018 21:37:59
```

Register dump:
```
--------------------------- GCLK
GCLK_MAIN:  GEN00 (always)
GCLK_RTC:  GEN02
GCLK_USB:  GEN00
GCLK_SERCOM3_CORE:  GEN00
GEN00:  DFLL48M OE
GEN01:  DFLL48M/150 OE
GEN02:  OSC32K/32
--------------------------- SYSCTRL
PCLKSR:  OSC32KRDY DFLLRDY DFLLLCKF DFLLLCKC BOD33RDY B33SRDY
OSC32K:  EN32K ONDEMAND STARTUP=92us CALIB=0x37
OSCULP32K:  CALIB=0x17
OSC8M:  ONDEMAND PRESC=8 CALIB=0x817 FRANGE=8-11MHz
DFLL:  MODE USBCRM CCDIS
BOD33:  ACTION=RESET PSEL=1 LEVEL=0x7
VREG:
VREF:  CALIB=0x60
--------------------------- RTC MODE0
CTRL:  PRESCALER=1024
READREQ:
EVCTRL:
FREQCORR:  0x80
COMP0:  0x0
```

Diff:
```diff
diff --git a/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h b/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h
index 5b9ec6ed6..74ec2bee9 100644
--- a/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h
+++ b/ports/atmel-samd/asf4_conf/samd21/hpl_sysctrl_config.h
@@ -109,7 +109,7 @@
 // <i> Indicates whether configuration for OSC32K is enabled or not
 // <id> enable_osc32k
 #ifndef CONF_OSC32K_CONFIG
-#define CONF_OSC32K_CONFIG 0
+#define CONF_OSC32K_CONFIG 1
 #endif

 // <h> 32kHz Internal Oscillator (OSC32K) Control
@@ -117,7 +117,7 @@
 // <i> Indicates whether Internal 32K Oscillator is enabled or not
 // <id> osc32k_arch_enable
 #ifndef CONF_OSC32K_ENABLE
-#define CONF_OSC32K_ENABLE 0
+#define CONF_OSC32K_ENABLE 1
 #endif

 // <q> On Demand Control
@@ -142,7 +142,7 @@
 // <i> Enable 32 Khz Output
 // <id> osc32k_arch_en32k
 #ifndef CONF_OSC32K_EN32K
-#define CONF_OSC32K_EN32K 0
+#define CONF_OSC32K_EN32K 1
 #endif

 // <q> Enable 1K
```

notro, 2018-05-08:
Fixed by #785

closed 2018-05-08 by notro