adafruit/circuitpython #778: I2C buffer length limited to 64 bytes transmitting

paulswirhun, 2018-04-25:
It appears I2C can only send out about 64 bytes at a time, and this is not documented as far as I can tell. I finally scoped my I2C bus with a Saleae when transmitting to the CY8C9560A (which requires a packet about 146-byte or more in length), and discovered that only the first 64 data bytes are sent by the Trinket M0.

Can some circuitpython developer confirm this, and maybe update the documentation to explain the limitation? If this is indeed a limitation, then it would be nice if <i2c_obj>.write() threw an exception explaining the buffer length limitation.

I'm using the adafruit slim wrapper of Circuitpython's I2C (from adafruit_bus_device import i2c_device), but I looked at the source code for that module and there doesn't appear to be any length limitation due to that wrapper.

tannewt, 2018-04-25:
What version of CircuitPython are you running?

paulswirhun, 2018-04-25:
>>> sys.version
'3.4.0'
(3, 4, 0)
>>> sys.implementation
(name='circuitpython', version=(2, 2, 4))

tannewt, 2018-04-25:
Thanks! Mind trying Alpha 6? There is new code for I2C in it.

paulswirhun, 2018-04-25:
Hmm...I usually prefer to stick with stable builds so I'm not dealing with platform issues. But, maybe I'll try upgrading and see if other things break. Is it not apparent from the Circuitpython C firmware what's going on? I also noticed that reads of > 64 bytes take a little pause at 64-byte boundaries, presumably to resize a buffer. Probably that wasn't implemented for transmit.

tannewt, 2018-04-25:
We're hard at work finishing 3.x so I'd rather see if its an issue there. Its pretty stable already. We're just missing `touchio` and `microcontroller.cpu.temperature`.

paulswirhun, 2018-04-26:
Ok, so I decided to try it out. Actually upgrading / downgrading is pretty easy thanks to the drag-and-drop method you all created. It took a few minutes longer because I had to also rebuild mpy-cross...I'm compiling everything to .mpy files to reduce memory usage; I even had to rewrite the ssd1306 driver to reduce the memory usage because the Trinket M0 was running out of memory with Adafruit's graphics-oriented version.

Long story short, the I2C write limitation is still there in 3.0.0 alpha 6. I'm attaching two Saleae analyzer captures. The data is 149 bytes long (command, command value, 146 byte payload, 1 CRC byte). The read works fine, and I think the pause in the middle is actually due to the Cypress CY8C9560A crossing a 64-byte boundary and needing to fetch more data (the Trinket M0 ACKs immediately). The write, however, just zeros out after the first 64 bytes. I'm fairly sure this is not the slave device, because it ACKs on every byte, and actually it looks as if the SCL is not clocking after the first 64 bytes. If it was a master/slave contention issue, I would expect SDA to be pulled low, like a slave ACKing when it should not.

[image]
[image]

As a further investigation, I tried splitting up the write in two different ways:
1) using start/end = 0/64, 64/128, 128/149
2) allocating 3 separate bytearrays with 64-byte chunks and sending them in sequence with stop=False/False/True

Neither of these work, and I still get the same result (zeroed data after 64 bytes and OSError: 5).

tannewt, 2018-04-26:
Thank you so much for the confirmation! Will add it to the 3.0 milestone so we fix it before the stable release.

paulswirhun, 2018-04-26:
Ok thanks for looking into this.

paulswirhun, 2018-04-26:
The Cypress datasheet actually comments "If current address cross 64-byte bloc boundary, then device performs real writing to EEPROM", so another possibility is that it's a timeout issue in Circuitpython, and the I2C library gets screwed up waiting a long time for an ACK. The EEPROM programming on that part takes < 100ms at above-freezing temperatures, per the datasheet, and I see that the 64th data byte is ACK'd after waiting about 52.7ms, presumably it was writing the EEPROM during that time.

paulswirhun, 2018-04-26:
Aha, sorry to spam you with updates. Feel free to ignore until you have time to address this bug. That Cypress part says "The I2C slave in this device requires that the I2C master supports clock stretching", which is basically that the slave holds SCL low while it performs slow operations like programming flash memory. I think this is what's happening in the 53ms when SCL is low and SDA is high, and it clearly screws up the Trinket M0.

Looking up the [ATSAMD21E18 datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/40001882A.pdf), there is an errata entry which says

"
1 – The I2C Slave SCL Low Extend Time-out (CTRLA.SEXTTOEN) and
Master SCL Low Extend Time-out (CTRLA.MEXTTOEN) cannot be used
if SCL Low Time-out (CTRLA.LOWTOUT) is disabled. When
SCTRLA.LOWTOUT=0, the GCLK_SERCOM_SLOW is not requested.
Errata reference: 12003
Fix/Workaround:
To use the Master or Slave SCL low extend time-outs, enable the SCL Low
Time-out (CTRLA.LOWTOUT=1).
"

I bet that's the issue!

tannewt, 2018-04-26:
Interesting! Does CircuitPython throw an Exception once it times out? We should if we can't write the whole buffer.

paulswirhun, 2018-04-26:
Circuitpython throws an OSError: 5. The sequence of events is:
- Master sends byte and waits for ACK
- Slave holds SCL low for a while (datasheet says < 100-200ms depending on temperature)
- Slave ACKs the last byte
- Master is now in a messed up state and sends a spurious connect-to-device-on-bus command when it should not.

Can you disable that timeout interrupt? Or, perhaps point me in that direction and I can build my own copy of Circuitpython with that modification? I did a little searching and think it would be in the 'busio' area of the SAMD21 area of this repo.

dhalbert, 2018-04-26:
It would be in `ports/atmel-samd/common-hal/busio/I2C.c`. But then you'll be getting into the ASF4 library. If you want to try building it yourself, see this draft guide on building: https://learn.adafruit.com/building-circuitpython/introduction?preview_token=bWdsAiUq9L8RNRzWikZZ2g. We can also make a test build for you with that bit set. I'm out for a couple of hours but will check back in later.

paulswirhun, 2018-04-26:
Thanks Dan,

I have a bit of time to dig through this, so no worries if this isn't a priority for you. I see where some defaults (interrupts disabled) are defined:
//ports/atmel-samd/asf4{_conf}/samd{21|51}/hpl_sercom_config.h

...but since these defines, such as CONF_SERCOM_1_I2CM_SEXTTOEN, appear nowhere else in the repo or its submodules, I'm not sure what those defines are doing.

I do see some "SEXTTOEN" appearing in these files:
//ports/atmel-samd/asf4/samd21/include/component/sercom.h
//ports/atmel-samd/asf4/samd21/hri/hri_sercom_d21.h

And in the first of those files it appears to enable the interrupt by writing '1'. It looks like there are a number of helper functions defined in the second file:

hri_sercomi2cm_{write|set|get|clear}_CTRLA_SEXTTOEN_bit

...however none of those functions are used anywhere, and the sercom.h file appears to just write the bit position manually.

paulswirhun, 2018-04-26:
Ok, I solved it! The problem is actually not deep inside the bit-level functions; the problem is a software timeout in //ports/atmel-samd/asf4/samd{21|51}/hpl/sercom/hpl_sercom.c

On line 1532 in the function _sercom_i2c_sync_wait_bus, they are using:
```C++
uint32_t timeout = 65535;
```
However, this is not an accurate way to do a timeout since it is just software-decremented in a loop of unknown execution time. I tried out progressively larger values (8x larger was not enough, so I went to 64x larger), and the problem is now resolved.

The only change necessary is to change that line to:
```C++
uint32_t timeout = 4194303;
```

dhalbert, 2018-04-27:
Very nice!

If we turn on the cache for SAMD51 (#783), the loop will take a shorter time, so it should count actual time instead of just counting up to a number. Do you have an idea of how many msecs should it be waiting?

paulswirhun, 2018-04-27:
@dhalbert: the Cypress part I'm using says it takes <200ms, so maybe it should be set to 250ms if you can time it accurately. However, if you're just going to include the hacky fix of increasing the software timeout value, it should be conservatively larger because the loop duration is not well controlled (it could change with library changes, compiler optimization flags, oscillator frequency, etc). If you decide to go with the software version, it probably warrants some kind of tick/tock measurement to see how that count value correlates to real time.

@tannewt: sounds like Dan might want a more thorough fix, so I'll leave it to you all to update; after all my change is only a few characters. (Also, although I use Git/Gerrit a lot at work, I haven't used Github so I have no idea how to submit a pull request).

@ladyada: Thanks! (what an honor :)

kattni, 2018-04-27:
@paulswirhun I'd be happy to walk you through the steps to do a pull request. It's only a few more steps beyond git, especially since you're already familiar with git. While we may eventually want a more thorough fix, the change you've proposed would still be an excellent addition, and it would be a great opportunity for an easy change to learn how to do pull requests.

Are you on the Adafruit Discord? We have a CircuitPython channel where we chat all the time. Visit adafru.it/discord to join, if you'd like to. Then we'd also have the option of real-time chat to help you through the steps. :)

paulswirhun, 2018-04-30:
Sure, I've joined but haven't figure out how to push to asf4 for review (so far it seems needlessly complicated, like all things git).

dhalbert, 2018-04-30:
Push the change to your own forked repo of asf4, in its own branch (a bugfix/enhancement branch), which you can delete later, after the PR is accepted. Then go to adafruit/asf4 in github and click "New pull request". If you have recently pushed, you will actually see a hint to do this. Set the branch to merge into to be `adafruit:circuitpython`. Then fill out the rest of the form and create the pull request.

paulswirhun, 2018-04-30:
Thanks, figured it out. I was trying to do it directly in the asf4 repo, but didn't have permissions I guess. I submitted it in my own fork now.

dhalbert, 2018-05-24:
@paulswirhun Since you tried this change of increasing the count, we've turned on the instruction/data cache on SAMD51, which may make the timeout loop now run too fast again. Could you try it again with 3.0.0-beta.0 or the tip of master?

tannewt, 2018-05-24:
Neither 3.0.0-beta.0 or master have the asf4 change so you'll need to build
a custom version to test the increased number.

paulswirhun, 2018-05-25:
I only have SAMD21 (Trinket M0).

tannewt, 2018-05-25:
3.0.0 is available on the Trinket M0 too.

PR #925 (2018-06-12): Update to the latest ASF4 which includes an I2C timeout extension …
while waiting for clock stretching.

Fixes #778
commit 73dadb06 (2018-06-12): Update to the latest ASF4 which includes an I2C timeout extension while waiting for clock stretching.

Fixes #778
closed 2018-06-12 by dhalbert