adafruit/circuitpython #781: PyCon US 2018!

tannewt, 2018-04-26:
@dhalbert, @kattni and I will be attending PyCon US 2018. Lets plan events here. We also have a dedicated [Discord](https://adafru.it/discord) channel, #pycon-us.

I ordered 500 business cards earlier today:
[image]
[image]

Please post other PyCon related plans here.

kattni, 2018-04-26:
We had a discussion today about what projects we want to have available. The plan is to have three tables with printouts setup and the projects are: capacitive touch, NeoPixels, and CPU temperature logging. Only one requires external HW, and we'll have that covered. 

I've ordered USB cables, alligator clips and NeoPixels. 

Many people will have USB cables already, but with Apple and USB-C formats becoming more prevalent, we wanted to have something available in case anyone didn't have a micro-USB option available. 

I will be soldering alligator clips to the NeoPixels so they're ready to use easily with the Gemmas. We're going to have 8 available for use (2 rings, 2 jewels, and 4 sticks - which was decided by stock levels, not a specific reason on our part). This will be the next step in preparation once the order arrives (likely next week).

tannewt, 2018-05-01:
Business cards shipped and should be to my place on Thursday.

kattni, 2018-05-01:
Ordered Adabox 006 for the PyLadies auction, as well as NeoPixel rings as they were in stock. Will arrive tomorrow. Alligator clips and USB cables are already here. Prep will likely be happening next week.

kattni, 2018-05-02:
Adabox 006 arrived today with the rest of the stuff we ordered for PyCon. Next week I'll be getting everything ready!

kattni, 2018-05-08:
Today I hacked up a bunch of alligator clips, stripped and tinned the cut ends, fed them through shrink wrap, and soldered them to eight NeoPixel rings.

[image]

They're now packed up with the rest of the stuff we plan on bringing with us. So that part is ready to go!

dhalbert, 2018-05-09:
1. I am experimenting with making single-page (probably two-sided) hardcopy cheat sheet, summarizing a bunch things and including the basic examples that will work on the Gemma.

2. Also may make up a gemma-pycon.uf2, with 2.3.1 and a filesystem containing stuff similar to what's shipped from the factory. Then people could update or restore easily.

kattni, 2018-05-17:
PyCon was an absolute blast and a complete success. Our Open Spaces were full and people were incredibly engaged. The sprints resulted in tons of updates to libraries, core code and documentation, and continued after we had already departed the conference. We gave the NeoPixel rings and other hardware we had to individuals who were absolutely excited to have them. One of the best moments was when people came to us with new ideas and projects outside the quickstart we created and were so incredibly excited to share with us. I know we'll still manage to improve on it next year.

Thanks everyone!

closed 2018-05-17 by kattni