adafruit/circuitpython #782: Verify drivers correctly import `struct`

tannewt, 2018-04-27:
In 2.x `struct` is available as `ustruct`. We need to make sure all drivers included in the bundle will work with 3.x which has renamed it to `struct`.

For backwards compatiblity it should be:

```python
try:
    import struct
except ImportError:
    import ustruct as struct
```

jepler, 2018-05-06:
I've grepped around in the circuitpython master branch source tree, its submodules, and Adafruit_CircuitPython_Bundle and its submodules and I didn't find any further problems.  There are at least a few which do the reverse of what is suggested, though, such as `ports/esp8266/modules/ntptime.py`:
```
try:
    import ustruct as struct
except:
    import struct
```
@tannewt is there somewhere else I should look, or should I go ahead and close this issue up?

tannewt, 2018-05-07:
The reverse is fine in ESP8266 modules. I'll close now. Thanks!

closed 2018-05-07 by tannewt