adafruit/circuitpython #787: Clock stretching not working with bitbangio?

caternuson, 2018-04-30:
Not 100% sure there is an issue here, but based on this:
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/9#issuecomment-384137352
it seems like something is not working when using bitbangio with an I2C device that clock stretches.

jerryneedell, 2018-05-13:
Should we raise an error if there is a closck-stretching timeout? Something is wrong if it times out.

I have not been able to produce any timeouts - I am getting some errors writing to the CCCS81 but that is nothing new :-(  I did find it works much better if I lower the I2C frequency from the default 400KHz to 100KHz or even down to 10KHz.

tannewt, 2018-05-13:
Yeah, please raise a `TimeoutError`. Thanks @jerryneedell !

tannewt, 2018-05-22:
@jerryneedell Is this done now?

jerryneedell, 2018-05-22:
The clock stretching timeout is in so I think the intent is complete. We may want to start a new issue for ongoing problems with bitbangio. @carternuson issue is not resolved but it may not be related to clock stretching.

caternuson, 2018-05-22:
Let's close it for now. We can re-open if it starts looking like clock stretching actually is the issue.

closed 2018-05-22 by caternuson

kevinjwalters, 2018-05-25:
I have been looking again at a problem where Gemma M0 using CircuitPython cannot talk to a STM32L-based tinyLiDAR over i2c. I've been using the pins which I believe allow it to use a hardware optimised driver.

Is there a difference between the software implemented i2c and that in the ATSAMD21? What version of CircuitPython would be best to try this with if this is the case? Can I just change pin to a non-hardware capable one and the s/w implementation will kick-in?

Discussion of problem I observed which still occurs: https://forums.adafruit.com/viewtopic.php?f=60&t=129362

dhalbert, 2018-05-25:
If you use `bitbangio.I2C`, you'll be using the software version. `busio.I2C` uses the hardware. So try `bitbang` if you have not already and see what happens.

caternuson, 2018-05-25:
Unfortunately, for non-Express boards like the Gemma M0, `bitbangio` is not baked in:
https://circuitpython.readthedocs.io/en/2.x/shared-bindings/index.html#support-matrix

kevinjwalters, 2018-05-30:
Yep, i was just joining things up in case there was some link between this and my issue.

My main problem turned out to be me not respecting the tinyLiDAR's wish (requirement!) for a pause between i2c commands, read/write/pause in a loop works ok. I think I got surprised by this because it seems to go non-recoverable rather than failing a few times plus I'm confused because I had tried some pauses before...

Speculating, there's a minor possibility this would have worked or been easier to diagnose with a documented and configurable i2c master read timeout.