adafruit/circuitpython #803: esp8266 port missing time.{sleep_us, sleep_ms}

larsks, 2018-05-03:
I wanted to use [this tm1637 module](https://github.com/mcauser/micropython-tm1637/) with CircuitPython, but it requires the `utime.sleep_us` method.  I see this reference in `modutime.c` in the CircuitPython sources, but it doesn't appear to be available on the `time` module once the board boots.

sommersoft, 2018-05-03:
So, I should have looked before, but I was pretty confident there are some timers available...

[`delay_us`](https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/common-hal/microcontroller/__init__.c#L45) is available from `microcontroller`. There are some additional delay/tick functions available at the C level in `esp_mphal`...

larsks, 2018-05-03:
@sommersoft, thanks for the feedback.

I see that `microcontroller.delay_us` is available, but it doesn't seem to work.  That is, I would expect `microcontroller.delay_us(1000000)` to delay for 1 second, but it returns immediately.  It's defined in terms of `os_delay_us()`, which I can't find anywhere.

larsks, 2018-05-04:
I have the `utime` module working now; the necessary changes are [here](https://github.com/adafruit/circuitpython/compare/master...larsks:feature/utime?expand=1).  I would be happy to submit this as a pull-request if it's the sort of thing people would like, but I'm not sure what CP thinks about the `u*` modules in general...

dhalbert, 2018-05-04:
We're discussing in issue #519 about how and where to add more time functionality. @tannewt has suggested adding a `precise_time` module for functionality above and beyond `time`.

We are trying to move away from the `u*` modules in favor of more descriptive ones with a more consistent API. We either provide a subset of or the same functionality provided by a CPython module, or we come up with a new module that has its own API, not necessarily derivative in terms of naming

larsks, 2018-05-04:
@tannewt while I understand your comments about timing accuracy, I will note that with the `sleep_us` function available, CircuitPython has no problem [driving the TM1637 display](https://youtu.be/CIjDf-0z9d8).

Either the accuracy is "good enough" or the TM1637 is really lenient.

tannewt, 2018-05-04:
Devices that you clock yourself like the TM1637 are very flexible. You just
can't clock them too fast. However, clocking them slower makes the update
speed slower. They'll still work though. That's why I was suggesting
removing the sleep all together. :-) I doubt CircuitPython on the SAMD21
can bitbang faster than the 2.5 mhz maximum for the chip.

The SAMD51 might be able to though. I haven't measured it. In that case, a
delay is helpful to make sure it doesn't go too fast.

tannewt, 2018-05-07:
We won't be adding any extra functions to `time` that aren't in CPython. #519 covers CircuitPython specific options so closing this issue.

closed 2018-05-07 by tannewt