adafruit/circuitpython #807: Plz Add Support for SAMD21 Internal Timer

rafi021, 2018-05-06:
I want to measure AC line frequency with circuit python and also want to interface with PZEM004T module with it. Please give me some suggestions of guideline for it. 
1. Need Internal Timer support of frequency measurement and a easy to use library. 
2. A library for PZEM004T module for measuring AC line Voltage, Current, Power and Energy.

sommersoft, 2018-05-06:
@rafi021 Thanks for the request! 

This is being looked into, due to related issue #516.

sommersoft, 2018-06-11:
@rafi021, I decided to separate the frequency counter part of this issue. I have working code for the SAMD, but it still needs polish and documentation.

tannewt, 2020-03-12:
@rafi021 What is the frequency range you are trying to measure?

dhalbert, 2025-01-31:
There is a MicroPython library: https://github.com/jacopoRodeschini/PZEM-004T/

closed 2026-05-16 by dhalbert