adafruit/circuitpython #809: NRF52 build does not start REPL

jerryneedell, 2018-05-07:
I just build the current master for the feather52 (nrf52)  
It builds within errors and I am able to flash it to the board with no issues, but I cannot access the REPL as I usually do via /dev/ttyUSB0 -- my screen session connects, but there is no response/
I also tried accessing the board via ampy, but it just hangs - no response.

jerryneedell, 2018-05-07:
It works normally if I revert to CP 3.0.0-alpha.6

jerryneedell, 2018-05-07:
This issue is still present after #811. 
No response  from REPL in terminal session or via ampy.

I can still make it work by removing the define for CIRCUITPY_BOOT_OUTPUT_FILE in mpconfigport.h

jerryneedell, 2018-05-07:
FYI - I noted that the ESP8266 mpconfigport.h does not define CIRCUITPY_BOOT_OUTPUT_FILE  and it continues to function normally.  Is it intentional that it not be defined and boot_out.txt not be created for the ESP8266?  Should it also be this way for the nrf52?

tannewt, 2018-05-07:
The nrf52 build should model itself after the SAMD port. The ESP8266 version isn't using the shared supervisor code yet.

microbuilder, 2018-05-14:
Just saw this (no email, not sure why, sorry) ... I'll see if I can reproduce.

jerryneedell, 2018-05-14:
@microbuilder #813  provides a workaround  - Ithaves been meager in to master so the nrf52 does boot to REPL now  -- the long term fix is still in progress.

hathach, 2018-05-15:
the issue is caused by the **mp_hal_stdout_tx_str**() is called before **serial_init**() is invoked due to **CIRCUITPYTHON_BOOT_OUTPUT_FILE**. The fix is skip output to serial if it is not inited.

https://github.com/microbuilder/circuitpython/commit/53593d1315e589416c78d23cfe7d021d6d1443e8

tannewt, 2018-05-16:
@hathach Thank you for the fix. Please make a pull request with the change and we'll merge it in.

tannewt, 2018-07-10:
I think this is fixed.

commit 53593d13 (2018-06-15): fix #809

define CIRCUITPY_BOOT_OUTPUT_FILE will cause mp_hal_stdout_tx_strn() to
invoke before serial_init() is called. Solution is skipped output to
serial if it is not inited.
closed 2018-07-10 by tannewt