adafruit/circuitpython #823: samd51 DigitalInOut always reports pull as None

deshipu, 2018-05-09:
```python
>>> import digitalio
>>> import board
>>> pin = digitalio.DigitalInOut(board.D4)
>>> pin.pull
>>> pin.switch_to_input(digitalio.Pull.UP)
>>> pin.pull
>>> pin.pull = digitalio.Pull.UP
>>> pin.pull
```

dhalbert, 2018-05-10:
This is weird: it's pin-dependent:
```
>>> import board,digitalio
>>> for name in dir(board):
...     try:
...         pin = digitalio.DigitalInOut(getattr(board,name))
...         pin.switch_to_input(digitalio.Pull.UP)
...         print(name, pin.pull)
...     except:
...         print("skip", name)
...         
...         
... 
A0 digitalio.Pull.UP
A1 digitalio.Pull.UP
A2 digitalio.Pull.UP
A3 digitalio.Pull.UP
A4 None
A5 None
D0 digitalio.Pull.UP
skip RX
D1 digitalio.Pull.UP
skip TX
D2 None
D3 None
D4 None
D5 None
D6 None
D7 None
D8 digitalio.Pull.UP
D9 digitalio.Pull.UP
D10 digitalio.Pull.UP
D11 digitalio.Pull.UP
D12 digitalio.Pull.UP
D13 digitalio.Pull.UP
SDA None
SCL None
AREF digitalio.Pull.UP
NEOPIXEL None
SCK digitalio.Pull.UP
MOSI digitalio.Pull.UP
MISO digitalio.Pull.UP
skip LED_RX
skip LED_TX
```

jerryneedell, 2018-05-10:
FYI - I was working on an ItsyBitsy M0 Express so I tried it there:
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.0.0-alpha.6-67-ge4ec672-dirty on 2018-05-09; Adafruit Itsy Bitsy M0 Express with samd21g18
>>> import board,digitalio
>>> for name in dir(board):
...     try:
...         pin=digitalio.DigitalInOut(getattr(board,name))
...         pin.switch_to_input(digitalio.Pull.UP)
...         print(name,pin.pull)
...     except:
...         print("skip",name)
...         
...         
... 
D0 digitalio.Pull.UP
skip RX
D1 digitalio.Pull.UP
skip TX
D2 digitalio.Pull.UP
D3 digitalio.Pull.UP
D4 digitalio.Pull.UP
D5 digitalio.Pull.UP
D6 digitalio.Pull.UP
D7 digitalio.Pull.UP
D8 digitalio.Pull.UP
D9 digitalio.Pull.UP
D10 digitalio.Pull.UP
D11 digitalio.Pull.UP
D12 digitalio.Pull.UP
D13 digitalio.Pull.UP
skip L
A0 digitalio.Pull.UP
A1 None
A2 None
A3 digitalio.Pull.UP
A4 digitalio.Pull.UP
A5 None
MOSI None
MISO digitalio.Pull.UP
SCK None
SCL digitalio.Pull.UP
SDA digitalio.Pull.UP
APA102_MOSI digitalio.Pull.UP
APA102_SCK digitalio.Pull.UP
>>> 
```

PR #827 (2018-05-11): PBxx pins were not setting DRVSTR or reporting PULL properly
Forgot to mask pin number for certain hri calls. PBxx pins were not getting DRVSTR set to strong properly or reporting their pull properly. 

Fixes #823
closed 2018-05-11 by tannewt