adafruit/circuitpython #825: Flashing Circuitpython to NRF with different SWD Programmer

No1089, 2018-05-10:
I’d like to flash a NRF52 chip with Circuitpython but I don’t have the SEGGER, but can buy an Atmel ICE locally - is it possible to have the makefile just output the files to flash?

jepler, 2018-05-10:
I'm not familiar with Atmel ICE and I don't know whether it can program the nrf52 chip.

However, to answer the specific question you asked, when you `make all` in ports/nrf52, you will get the firmware in several formats, "elf", "bin" and "hex".  These are all "fairly common" formats, and at least one of them is likely to be supported by any flashing tool software.

sommersoft, 2018-05-10:
@No1089 As you may or may not have read, `nrfjprog` is designed to specifically work with the SEGGER; no indication that it works with any other SWD. I don't see any other utility for other debuggers/programmers, and google isn't turning anything up other than a couple questions.

You could look into programming it via Device Firmware Update (DFU) over serial, which is how the Feather52 - nRF52832 is programmed (other than via SEGGER).

tannewt, 2018-05-11:
@jerryneedell @microbuilder @arturo182 Would know too.

No1089, 2018-05-11:
@jepler @sommersoft I had a look at the makefile and as I understood it, it's all fine up to the point where the flasher is called. If the output file can be saved, a different piece of hardware can flash the chip (according to what I know). But I am not sure how the `make sd` part works to flash the bluetooth stack. Sorry for the ignorance, never used of of these programmers/debuggers before.

No1089, 2018-05-11:
@sommersoft I can source the sparkfun nRF52832 breakout locally, but I want to use this when it arrives: [](https://www.aliexpress.com/item/Fast-Free-Ship-NRF52832-Bluetooth-4-0BLE-SOC-Wireless-Sensor-Acceleration-Sensor-BME280-Environmental-Sensor/32818538078.html?spm=a2g0s.13010208.99999999.262.CScVLv).

deshipu, 2018-05-11:
As far as I know, Atmel ICE is just a locked down version of the Segger.

arturo182, 2018-05-11:
According to this http://omzlo.com/articles/programming-samd21-using-atmel-ice-with-openocd it looks like the Atmel ICE is a CMSIS-DAP programmer, so it should be possible to flash a nRF chip using it with the help of OpenOCD.

tannewt, 2018-07-10:
Looks like it should be possible.

closed 2018-07-10 by tannewt