adafruit/circuitpython #853: REPL paste mode (CTRL-E) crashing board to safe mode

ash0x1b, 2018-05-19:
I'm experiencing strange behaviour with REPL in paste mode. After an arbitrary period of time, pasting anything while in REPL paste mode will crash the board and bring it back up in safe mode.

While in safe mode, paste mode works without an issue. Resetting it back to normal mode, paste mode will crash it to safe mode every single time. Typing in 4 characters will cause a crash. The crash will occur while typing in the 4. character. CTRL-C or CTRL-D will cause a crash immediately.

Cycling power and removing libs and scripts from internal storage doesn't solve the issue. I have to, either reflash the board with CP firmware, use `flash_erase_express.ino.metro_m0.uf2` or run `storage.erase_filesystem()` from REPL.

I first suspected that the filesystem got corrupted during many resets. I've reflashed the board and made sure USB was ejected cleanly. Same issue occurred after two hours of running the script.

Board: Metro M0 Express
Libs: adafruit_dotstar

Script I was running as `code.py`
```python
import time
import random
import board
import adafruit_dotstar as dotstar

dots = dotstar.DotStar(board.SCK, board.MOSI, 50, brightness=0.01, auto_write=False)

######################### HELPERS ##############################

# a random color 0 -> 224
def random_color():
    return random.randrange(0, 7) * 32

######################### MAIN LOOP ##############################
n_dots = len(dots)
try:
        
    while True:
        #fill each dot with a random color
        for dot in range(n_dots):
            dots[dot] = (random_color(), random_color(), random_color())
        # show all dots in strip
        dots.show()

        time.sleep(.25)
except KeyboardInterrupt:
    dots.fill((0, 0, 0))
    dots.show()
```

tannewt, 2018-05-22:
Could you please provide step by step instructions on how I could reproduce this? You say you are using paste mode but also provide a code.py. Which one is running when the crash happens?

ash0x1b, 2018-05-23:
I've managed to reproduce it on [Adafruit Feather M0 Bluefruit LE](https://www.adafruit.com/product/2995) too.

Steps to take:
* Flash `adafruit-circuitpython-feather_m0_basic-2.3.1.bin` with `bossac`
* eject `CIRCUITPYTHON` and unplug USB
* plug-in USB
* connect to REPL
* enter paste mode (CTRL-E)
* start typing (123 is enough to crash the board)

When I get home I'll check if the same happens on Metro M0 express.

tannewt, 2018-05-23:
Thank you for the detailed instructions! I was able to reproduce it with 2.3.1. Now to find the bug. :-)

tannewt, 2018-05-24:
I've managed to get a backtrace for this. Its fixed by #849 and #848. Please use 3.0.0-beta.0.

```
(gdb) bt
#0  0x00020702 in HardFault_Handler () at asf/sam0/utils/cmsis/samd21/source/gcc/startup_samd21.c:285
#1  <signal handler called>
#2  0x00021910 in validate.lto_priv.645 (obj=obj@entry=0x20007dac, fs=fs@entry=0x20007d18) at ../lib/oofatfs/ff.c:3075
#3  0x000224b4 in f_write (fp=0x20007dac, buff=buff@entry=0x20007dac, btw=btw@entry=1, bw=bw@entry=0x20007d58) at ../lib/oofatfs/ff.c:3453
#4  0x0002488a in mp_hal_stdout_tx_strn (str=<optimized out>, len=<optimized out>) at mphalport.c:204
#5  0x00019dee in pyexec_friendly_repl () at ../lib/utils/pyexec.c:481
#6  main () at main.c:815
```

closed 2018-05-24 by tannewt