adafruit/circuitpython #856: gc.collect after import

ladyada, 2018-05-21:
in *theory* it shouldnt make a difference whether we collect() after each import in a sketch but ive found that "sketch that work but then fail after a while with memerrors dont have failures if we gc.collect after each import" im testing CPX in 3.0.0

we don't import a lot, its not a time sensitive event, and there's a huge memory slush...why not :)

tannewt, 2018-05-23:
I'd really like to investigate why this is the case after 3.0.0 is done. Do have an example handy that could be used? Thanks.

ladyada, 2018-05-23:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/Crickits/yayayay_flier/code.py

tannewt, 2018-06-27:
I'm looking at this today. I've got my heap analysis setup going.

commit 9de611c0 (2018-07-09): Always collect after an import

Imports generate a lot of garbage so cleaning it up immediately
reduces the likelihood longer lived data structures don't end up in
the middle of the heap.

Fixes #856
closed 2018-07-09 by dhalbert