adafruit/circuitpython #857: create driver for TMP007

jerryneedell, 2018-05-21:
I could not find a driver for the TMP007 IR Temperature sensor https://www.adafruit.com/product/2023
There is a Python Driver for the Raspberry Pi and I now have adapted it to work under CirCuitPython.
MY initial version seems to be functional.

@tannewt when you have a chance, please create a template for Adafruit_CircuitPython_TMP007

tannewt, 2018-05-22:
Thanks @jerryneedell ! New repo is here: https://github.com/adafruit/Adafruit_CircuitPython_TMP007

Please propose a PR and let me know if you need me to run cookiecutter.

jerryneedell, 2018-05-22:
In the past I have run cookie cutter locally then uploaded the files in the PR - Is that ok or is there a better way?

tannewt, 2018-05-22:
Thats ok!

ladyada, 2018-08-18:
driver exists, in bundle and was tested!

closed 2018-08-18 by ladyada