adafruit/circuitpython #874: 3.0 Beta: busio.UART read(nbytes) returns empty bytearray on timeout rather than None

CedarGroveStudios, 2018-05-25:
In 3.0.0-beta.0 2018-05-23, busio.UART read(nbytes) does not return the None value upon timeout, just an empty bytearray.

Tested using Metro M4 express hardware.

tannewt, 2018-06-12:
Will validate that it [matches CPython](https://docs.python.org/3.6/library/io.html):

```
 read(size=-1)

    Read up to size bytes from the object and return them. As a convenience, if size is
unspecified or -1, all bytes until EOF are returned. Otherwise, only one system call is
ever made. Fewer than size bytes may be returned if the operating system call
returns fewer than size bytes.

    If 0 bytes are returned, and size was not 0, this indicates end of file. If the object
is in non-blocking mode and no bytes are available, None is returned.
```

closed 2018-06-12 by dhalbert

CedarGroveStudios, 2018-06-12:
3.0 works differently than CPy 2.3.1. The examples break with 3.0 because of the bytearray content check:
`data = uart.read(1)  # get a byte from RX
    if data is not None:
        …`

tannewt, 2018-06-13:
Yup, it should work now @CedarGroveStudios

CedarGroveStudios, 2018-06-13:
Super! I'll test it on the next release. Thanks!

PR #928 (2018-06-12): Correct UART reads to return error on timeout.
This causes read to correctly return None instead of b''.

Fixes #874
commit ee817a2f (2018-06-12): Correct UART reads to return error on timeout.

This causes read to correctly return None instead of b''.

Fixes #874