adafruit/circuitpython #875: main.py shipped on Metro M0 for CircuitPython 2.2.0 runs out of memory

rhooper, 2018-05-26:
MemoryError: memory allocation failed, allocating 1016 bytes

After using the attached updater to update the existing libs to the May 25th bundle, I get a memory error.

```
Adafruit CircuitPython 3.0.0-beta.0-22-g754c36bb3 on 2018-05-26; Adafruit Metro M0 Express with samd21g18
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 14, in <module>
  File "adafruit_hid/keyboard.py", line 36, in <module>
  File "adafruit_hid/keycode.py", line 31, in <module>
  File "adafruit_hid/keycode.py", line 245, in Keycode
MemoryError: memory allocation failed, allocating 1016 bytes
```

main.py as received on the board
https://gist.github.com/rhooper/0cf8de8a386c79a6eec1109f230a24e1

updater script:
```bash
#!/bin/bash

MOUNTPOINT=/Volumes/CIRCUITPY

if [ ! -d $MOUNTPOINT ]; then
	echo "$MOUNTPOINT not found";
	exit 1
fi

(cd $MOUNTPOINT find lib -name \*.mpy) |
	xargs -n 1 -I {} cp -v {} $MOUNTPOINT/{}

sleep 3
find $MOUNTPOINT -name ._\* -exec  rm -v {} \;
```

rhooper, 2018-05-27:
Is there a repo that contains the initial CircuitPython images for the various hardware?  I couldn't find one, but think that would be a better way to track this than sharing around code files.  It would make it easier for people to do PRs and also see how the code evolves over time (eg 2.0 to 3.0 circuitpython)?

ladyada, 2018-05-27:
https://learn.adafruit.com/adafruit-metro-m0-express-designed-for-circuitpython/downloads

is where the files are now, there's no repo at this time the 3.0 code doesnt fully work either so we only want to place it here where you can read our warnings

ladyada, 2018-05-29:
ok ill remove HID from the example

ladyada, 2018-05-30:
here's a new fresh linted example! please try it :)
[metro 3.0.zip](https://github.com/adafruit/circuitpython/files/2052197/metro.3.0.zip)

rhooper, 2018-05-30:
Works fine for me on `Adafruit CircuitPython 3.0.0-beta.0-22-g754c36bb3 on 2018-05-29; Adafruit Metro M0 Express with samd21g18`!

Did not test: D2,D3,D4 buttons or audio out :)

ladyada, 2018-06-06:
ok i think this can be closed then?

closed 2018-06-06 by rhooper