adafruit/circuitpython #876: samd21: time.monotonic() slows down under pressure

notro, 2018-05-27:
I don't know if this is by (unfortunate) design or not, but time.monotonic() slows down under pressure.

I run a script to measure RTC accuracy and it slows down the monotonic clock.

This snippet is run on the board through the REPL to give me the moment that the second ticks over:
```python
import time

lt1 = time.localtime()
lt2 = lt1
i = 0
while lt1.tm_sec == lt2.tm_sec:
    lt2 = time.localtime()
    i += 1
m = time.monotonic()
print("%%s:%%.6f:%%d" %% ( str(tuple(lt2)), m, i ))

```

This is the resulting test script output:
```
-0.005966 secs / 0.000025 days = -238.187512 secs/day (-2756.80 ppm) : 27 May 2018 22:38:30 (2286.009 - 2285.562 = 0.447) : 127
-0.001031 secs / 0.000037 days = -28.206318 secs/day (-326.46 ppm) : 27 May 2018 22:38:31 (2286.456 - 2286.009 = 0.447) : 124
```
The important part is the last two numbers: monotonic has elapsed ~0.5 seconds and the loop has run 127/124 times.

If I add a 100ms delay in the loop:
```python
import time

lt1 = time.localtime()
lt2 = lt1
i = 0
while lt1.tm_sec == lt2.tm_sec:
    lt2 = time.localtime()
    i += 1
    time.sleep(0.1)
m = time.monotonic()
print("%%s:%%.6f:%%d" %% ( str(tuple(lt2)), m, i ))

```

Then I get that monotonic has elapsed ~1 second and the loop has run 7/8 times:
```
+0.055680 secs / 0.000026 days = +2114.643333 secs/day (+24475.04 ppm) : 27 May 2018 22:40:08 (2366.929 - 2366.016 = 0.913) : 7
+0.010000 secs / 0.000038 days = +260.184927 secs/day (+3011.40 ppm) : 27 May 2018 22:40:09 (2367.941 - 2366.929 = 1.013) : 8
-0.025755 secs / 0.000050 days = -510.800835 secs/day (-5912.05 ppm) : 27 May 2018 22:40:10 (2368.938 - 2367.941 = 0.997) : 8
+0.029896 secs / 0.000061 days = +487.286023 secs/day (+5639.88 ppm) : 27 May 2018 22:40:11 (2369.850 - 2368.938 = 0.911) : 7
-0.026185 secs / 0.000074 days = -355.902557 secs/day (-4119.24 ppm) : 27 May 2018 22:40:12 (2370.868 - 2369.850 = 1.019) : 8
```

This is with a 10ms sleep:
```
+0.006152 secs / 0.000025 days = +243.827366 secs/day (+2822.08 ppm) : 27 May 2018 22:56:36 (3344.759 - 3343.982 = 0.777) : 54
+0.001943 secs / 0.000037 days = +52.725427 secs/day (+610.25 ppm) : 27 May 2018 22:56:37 (3345.534 - 3344.759 = 0.775) : 55
+0.007818 secs / 0.000048 days = +161.654050 secs/day (+1871.00 ppm) : 27 May 2018 22:56:38 (3346.313 - 3345.534 = 0.779) : 53
```

Does anyone have an explanation for this?

notro, 2018-05-29:
The thing is, monotonic should always have elapsed one second or more, never (much) less. I'm using monotonic to measure the time it takes for one second to elapse on the RTC.

I have now run the same test on a Metro M4 Express, and it doesn't show the same problem.
This is with no sleeping in the loop: +700 loop iterations and ~1.0 second elapsed on monotonic:
```
+0.005894 secs / 0.000025 days = +234.605899 secs/day (+2715.35 ppm) : 29 May 2018 00:00:02 (56083.937 - 56082.935 = 1.001) : 767
+0.001815 secs / 0.000037 days = +49.392845 secs/day (+571.68 ppm) : 29 May 2018 00:00:03 (56084.952 - 56083.937 = 1.016) : 761
-0.002347 secs / 0.000048 days = -48.521483 secs/day (-561.59 ppm) : 29 May 2018 00:00:04 (56085.954 - 56084.952 = 1.001) : 770
-0.007010 secs / 0.000060 days = -116.838150 secs/day (-1352.29 ppm) : 29 May 2018 00:00:05 (56086.950 - 56085.954 = 0.997) : 759
-0.011479 secs / 0.000072 days = -160.283728 secs/day (-1855.14 ppm) : 29 May 2018 00:00:06 (56087.952 - 56086.950 = 1.001) : 768
-0.005442 secs / 0.000083 days = -65.464984 secs/day (-757.70 ppm) : 29 May 2018 00:00:07 (56088.953 - 56087.952 = 1.001) : 757
-0.009664 secs / 0.000095 days = -101.999383 secs/day (-1180.55 ppm) : 29 May 2018 00:00:08 (56089.969 - 56088.953 = 1.016) : 770
-0.014307 secs / 0.000106 days = -134.497285 secs/day (-1556.68 ppm) : 29 May 2018 00:00:09 (56090.965 - 56089.969 = 0.997) : 759
-0.018378 secs / 0.000118 days = -155.750179 secs/day (-1802.66 ppm) : 29 May 2018 00:00:10 (56091.967 - 56090.965 = 1.001) : 768
-0.012985 secs / 0.000130 days = -100.261177 secs/day (-1160.43 ppm) : 29 May 2018 00:00:11 (56092.968 - 56091.967 = 1.001) : 757
-0.017704 secs / 0.000141 days = -125.438034 secs/day (-1451.83 ppm) : 29 May 2018 00:00:12 (56093.969 - 56092.968 = 1.001) : 765
-0.022001 secs / 0.000153 days = -144.020932 secs/day (-1666.91 ppm) : 29 May 2018 00:00:13 (56094.985 - 56093.969 = 1.016) : 759
```

I can upload a test so people can try and reproduce when I've done with tools/cpboard.py, which the test depends upon.

tannewt, 2018-05-29:
I don't think that's what your code above is doing. I think its measuring the time until the first seconds change from script start up. I don't see where it waits to ensure the first monotonic is captured on a seconds change. Do you have an updated version of the code?

tannewt, 2018-05-29:
Ah, its possible that the USB code is starving the SysTick interrupt. Try a similar test that doesn't use USB actively while measuring monotonic vs the RTC.

dhalbert, 2018-06-12:
@notro could you retest? Thanks. @tannewt redid the interrupt priorities so that Systick is now at the highest prio instead of the lowest (!) as it was before.

notro, 2018-06-12:
I'm still getting the same result.

Here's a stripped down version of my example:

code.py:
```python
import time

lt1 = time.localtime()
lt2 = lt1
m1 = time.monotonic()
while True:
    i = 0
    while lt1.tm_sec == lt2.tm_sec:
        lt2 = time.localtime()
        i += 1
        #time.sleep(0.1)
    m2 = time.monotonic()
    print("time: %s monotonic:%.6f iterations:%d" % ( str(tuple(lt2)[:6]), m2 - m1, i ))
    lt1 = lt2
    m1 = m2
```

Result with no sleep:
```
time: (2000, 1, 1, 0, 8, 48) monotonic:0.017944 iterations:13
time: (2000, 1, 1, 0, 8, 49) monotonic:0.343994 iterations:153
time: (2000, 1, 1, 0, 8, 50) monotonic:0.310059 iterations:158
time: (2000, 1, 1, 0, 8, 51) monotonic:0.353027 iterations:150
time: (2000, 1, 1, 0, 8, 52) monotonic:0.354980 iterations:146
time: (2000, 1, 1, 0, 8, 53) monotonic:0.355957 iterations:149
time: (2000, 1, 1, 0, 8, 54) monotonic:0.322021 iterations:155
time: (2000, 1, 1, 0, 8, 55) monotonic:0.344971 iterations:150
```

Result with 100ms sleep:
```
time: (2000, 1, 1, 0, 9, 1) monotonic:0.902954 iterations:9
time: (2000, 1, 1, 0, 9, 2) monotonic:1.011963 iterations:10
time: (2000, 1, 1, 0, 9, 3) monotonic:0.911011 iterations:9
time: (2000, 1, 1, 0, 9, 4) monotonic:1.010010 iterations:10
time: (2000, 1, 1, 0, 9, 5) monotonic:0.910034 iterations:9
time: (2000, 1, 1, 0, 9, 6) monotonic:1.010010 iterations:10
```

notro, 2018-06-26:
I just tried the test case in the previous post on an M4 (without sleep) and the number of iteration varies quite a lot:
```
Adafruit CircuitPython 3.0.0-rc.0-25-gd7d132d5a-dirty on 2018-06-26; Adafruit Metro M4 Express with samd51j19
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
time: (2000, 1, 1, 0, 0, 39) monotonic:0.244003 iterations:403
time: (2000, 1, 1, 0, 0, 40) monotonic:1.003006 iterations:548
time: (2000, 1, 1, 0, 0, 41) monotonic:1.000000 iterations:1120
time: (2000, 1, 1, 0, 0, 42) monotonic:1.000992 iterations:396
time: (2000, 1, 1, 0, 0, 43) monotonic:0.999008 iterations:953
time: (2000, 1, 1, 0, 0, 44) monotonic:1.002991 iterations:387
time: (2000, 1, 1, 0, 0, 45) monotonic:1.000000 iterations:977
time: (2000, 1, 1, 0, 0, 46) monotonic:1.000000 iterations:372
time: (2000, 1, 1, 0, 0, 47) monotonic:1.001007 iterations:949
time: (2000, 1, 1, 0, 0, 48) monotonic:1.000992 iterations:371
time: (2000, 1, 1, 0, 0, 49) monotonic:1.000000 iterations:952
time: (2000, 1, 1, 0, 0, 50) monotonic:1.001999 iterations:370
time: (2000, 1, 1, 0, 0, 51) monotonic:0.998001 iterations:949
Traceback (most recent call last):
  File "code.py", line 9, in <module>
KeyboardInterrupt:

Press any key to enter the REPL. Use CTRL-D to reload.

```

Dar-Scott, 2020-01-15:
Perhaps stating the obvious... I am new to ARM, CP implementation, and even github... I am today seeing this same behavior. 
I looked at the systick handler. (I think it does some unneeded things; see below.) I don't see a problem. I didn't dive down into some functions such as one related to the file system. Perhaps those can be moved to software triggered exceptions (maybe using the STIR, if available, for an atomic trigger). Those can then run at a lower priority. From my limited glance, I don't think the handler run time is too long.
Since the systick handler now runs at highest priority (and presumably nothing does; this can be checked, I guess), the problem is not because of higher priority interrupts or bumping priority. Given those assumptions, I say **the loss of time is caused by the disabling of interrupts for long periods**. Perhaps a style guide is needed here, or, if there is one, attention to it is needed. Interrupts should be disabled for tight short times as needed, not for large operations. Priority shifting can be considered as an alternative.
(As far as those unneeded things in the systick handler... I don't think the control register needs to be read. And even if it does, the read is atomic, so it doesn't need to be wrapped with interrupt disable/enable. And since the systick runs at highest priority, the wrapping will, I think, have no behavior effect.)

Dar-Scott, 2020-01-16:
I tinkered with the test script and it seems that time.localtime() or time.time() takes about 4.46 ms and during its execution 3 tick interrupts seem to be lost.

Dar-Scott, 2020-01-16:
Oh, the 4.5 ms for time.time() and the 3 lost SYSTICKs are from testing on Feather M0 Express.

tannewt, 2020-07-14:
I'm going to assume this is fixed by the switch to using the RTC for timing. Please comment if it's not fixed.

closed 2020-07-14 by tannewt