adafruit/circuitpython #877: [docs] add example for reading battery on feather m0 express in circuitpy

ladyada, 2018-05-27:
@kattni i think you already wrote this code? goes here
https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/power-management
(its a shared page with feather m0 basic but the pinouts are the same)

kattni, 2018-05-27:
@ladyada It was part of another project. I cleaned it up. Is this what you're looking for?

```
import time
import board
import analogio

vbat_voltage = analogio.AnalogIn(board.D9)

def get_voltage(pin):
    return (pin.value * 3.3) / 65536 * 2

while True:
    battery_voltage = get_voltage(vbat_voltage)
    print("VBat voltage: {:.2f}".format(battery_voltage))
    time.sleep(1)```

ladyada, 2018-05-27:
yeah like that but you dont need the while()

kattni, 2018-05-27:
Ok I added it to the page. Should be good to go.