adafruit/circuitpython #879: cpx + crickit PDMin crashtest

ladyada, 2018-05-28:
@dhalbert  can you try running https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/Crickits/gear_tower/sound-motor.py
with a DC motor, see if you can replicate some crashes we saw? no REPL was attached, i think it was just a reboot

dhalbert, 2018-05-29:
I can get this to crash even without doing any motor setting in the main loop, so I think it's just a `PDMIn`. It randomly stops at some point. I've seen 81 times or a few hundred times go by before it hangs. I'm wiring up a PDM mic with a Metro M0 so I can debug easily.

tannewt, 2018-06-05:
This is no longer an issue right?

dhalbert, 2018-06-05:
Unfortunately, still is. I can provoke a hang by doing a filesystem operation.

dhalbert, 2018-06-05:
Simple test, on Metro M0 with a 3492 breakout. But you don't actually have a microphone connected to get the hang.
```
import audiobusio
import time
import board
import array

# Our microphone
mic = audiobusio.PDMIn(board.D1, board.D9, sample_rate=16000, bit_depth = 16)
samples = array.array('H', [0] * 200)

i = 0
while True:
    i += 1
    num_recorded = mic.record(samples, len(samples))
    print(i, num_recorded)
```

To provoke a hang, do something some filesystem operation that writes CIRCUITPY.

dhalbert, 2018-06-05:
The symptom is that the loop around `common-hal/audiobusio/PDMIn.c:397` (`MICROPY_VM_HOOK_LOOP`) never exits.

In the older 2.x code, this line is in `start_dma()`:
```
    I2S->DATA[1].reg = I2S->DATA[1].reg;
```
Writing the register provokes the transfer to start, according to the datasheet.
If I add a similar line around the `// Record` line in `PDMIn.c`, but with the subscripts changed to `[0]`, then I get a worse crash: I get a hard fault in the MSC code, and the ASF4 device structs for MSC look like they've been overwritten.

So I'm thinking maybe there's some buffer overflow going on somewhere, and it can manifest in different ways.

dhalbert, 2018-06-05:
I tried essentially the same program on a Trinket M0 to rule SPI flash presence in or out. Had to hack things up a bit to provide I2S on a Trinket, but I get same kinds of hangs (not examined in gdb). Not clear if coincident with filesystem operations: sometimes seems synchronized, and sometimes operation does not provoke a hang. Often hangs by itself (more quickly than on Metro M0).

dhalbert, 2018-06-06:
I commented out escaping to the VM while filling the before, hoping that one of the background routines might be the issue. But I get the same kinds of hangs still :frowning_face: 
```
        while (!event_interrupt_active(event_channel)) {
            #ifdef MICROPY_VM_HOOK_LOOP
            // COMMENTED OUT                MICROPY_VM_HOOK_LOOP
            #endif
        }
```

dhalbert, 2018-06-06:
In addition to removing the `MICROPY_VM_HOOK_LOOP`, as described in the previous comment.  I wrapped the entire insides of `common_hal_audiobusio_pdmin_record_to_buffer()` in disable/enable interrupts, so it would be completely synchronous, and it still hangs the same way. When a hang happens, the first word of `first_buffer` contains good data, but the rest has not been filled in. So only one word was transferred.

Also checked settings of I2S peripheral. The settings are the same when a good transfer is happening as when as hang has happened.

dhalbert, 2018-06-08:
Per the previous comment: The DMA transfer is triggered, per beat, by the I2S peripheral reporting that there's new data. Somehow a trigger fails to show up at the very beginning of a DMA, and then the DMA transfer stalls after just one word has been transferred.

tannewt, 2018-06-08:
I'm actually re-opening because the root cause hasn't been found. I'll adjust the milestone though so this isn't a blocker.

PR #916 (2018-06-08): Check for PDMIn DMA getting stuck.
The DMA transfer in PDMIn occasionally seems to lose a trigger from the I2S peripheral, and stops transferring. PDMIn record would hang waiting for DMA to finish. This puts a counter "timeout" on the number of times we check for DMA not being done. If the counter is exceeded, we give up and try again.

It would be nice to know _why_ the trigger is being lost, but this prevents a hang, which is a lot better than nothing.

Sorta fixes #879.
PR #927 (2018-06-12): PDMIn: turn serializer on when starting DMA
Thought about this some more, and realized that synchronizing the turning on of the serializer with turning on DMA might solve the problem of lost triggers. The PDMIn clock to the microphone still runs all the time, to avoid microphone startup time (verified with Saleae), but the serializer is turned on and off as needed.

Tested on Metro M0 and M4, and works. I left in the hang prevention added in #916, because it's cheap, but tested with that temporarily turned off to make sure this new fix really does something. Rewrote #916 slightly to make it mo [... cut]
closed 2018-06-12 by tannewt