adafruit/circuitpython #883: update crickit guide to 'match' the CPX quickstart

ladyada, 2018-05-29:
this page is a little barebones :)
https://learn.adafruit.com/adafruit-crickit-creative-robotic-interactive-construction-kit/circuitpython-code

@kattni please copy over the side2side elements and pix from
https://learn.adafruit.com/adafruit-circuit-playground-express/circuitpython-quickstart

into the crickit BUT KEEP THE SPECIAL UF2! no new screenshots are needed, we'll have the same pix/animations, just a different UF2.

kattni, 2018-05-30:
Ok, excellent. Will do!

kattni, 2018-06-05:
Duplicated the CPX CP quickstart install into the Crickit Guide. Only changed the name of the UF2 referred to in one of the side2s to match the special UF2 for crickit.

closed 2018-06-05 by kattni