adafruit/circuitpython #884: (Large numbers of) Neopixel and Dotstar are slow

rhooper, 2018-05-29:
Both Neopixel and Dotstar take a lot of cycles to update pixels.  There's two different things going on that slow down both.  The first is that converting from Int or RGB[W] to the bytearray is complex code that makes updating pixels slow.  The penalty for more pixels is O(N) (as one would expect).

The second issue is that when brightness<1, extra work is done during `.show()` - the bytearray is copied and every pixel value recomputed.  This probably also causes unnecessary garbage collection, and has helped me run out of memory later on in runtime through apparent fragmentation.

This technically isn't a core CircuitPython issue, but because it requires a C helper to address, some work is likely needed in CircuitPython to add helpers to speed up these tasks.

rhooper, 2018-05-29:
I have done some benchmarking to understand the penalty of using brightness < 1 is significant, especially with larger strips or matrices (the penalty is noticeable even with a single pixel).

When testing 30, 60 and 72 pixel devices, the speedup from using a simple CircuitPython C helper to update the bytearray members was about 5x.  Speeding up `.fill()` could be even more significant, and a helper to do brightness conversions could be equally significantly faster, especially with a preallocated buffer.

rhooper, 2018-05-30:
I think I know what I want to do for speeding up NeoPixel and DotStar` _setitem()` - a pixelbuf class that takes over for (most of) the python bytearray management... specifically handling byte ordering and bpp=3/4 (RGB vs RGBW) itself.  It would also probably handle `.brightness` and do the bytearray brightness calculations itself ... or keep two bytearrays around (optional because of memory) - one with brightness applied and one with original values, updated during setitem.  A helper should exist to return an RGB[W] tuple back to the python code during `__getitem__`.

tannewt, 2018-05-31:
Sounds good. We can play around with it. Thanks!

kevinjwalters, 2018-05-31:
There's some previous discussion on this on: https://forums.adafruit.com/viewtopic.php?f=60&t=125609&p=630394#p630394

As mentioned, I ended up breaking the encapsulation and writing directly to the array for a bit more speed. I didn't feel too dirty as it was toy code but it's not ideal!

rhooper, 2018-12-27:
@ladyada Later on after today's call a number of us got stuck in a loop about the best way to name and release the pixelbuf work.  It was felt that you would have preferences as to what approach to take.

Option 1 is to have a new set of libraries that don't try to be backwards compatible with neopixel.py and adafruit_dotstar.py instead focussing on an API that is mostly compatible but designed for speed.

Option 2 is to do everything possible to make it completely backwards compatible, so that the new libraries are drop-in replacements for the old ones.

Option 3 is some other combination. 

My WIP current code uses these names/class hierarchy and repos/branches:

CircuitPython - https://github.com/rhooper/circuitpython/tree/pixelbuf-resync
Class: `pixelbuf.PixelBuf`
Also has `pixelbuf.ByteOrder` classes and subclasses for most byte orders, including things like `LRGB` and variants (for luminance, RGB - eg Dotstar).

RGBLED Library - This one is the one we weren't sure about the naming of, because it works for RGB, RGBW, BGR, LRGB, white only, etc.
https://github.com/rhooper/Adafruit_CircuitPython_RGBLED

My current working approach is to have hardware implementation subclasses in live in this repo, for lack of a better place to put them.  There was some concern later on the weekly call about having their names be the same as the existing `neopixel` and `adafruit_dotstar` ones, especially if we don't maintain backwards compatibility.

Class hierarchy, as the code stands as of writing this:

`pixelbuf.PixelBuf` - CircuitPython C class
`rgbled.RGBLED` - Abstract CircuitPython class subclassing `pixelbuf.PixelBuf`  implementing initialization helpers, context managers, repr, etc.
`neopixel.NeoPixel` - CircuitPython subclass of RGBLED that adds hardware specific support for dotstar.

As implemented right now, there are differences in the APIs, notably the ByteOrder classes and what properties exist on the object (eg `n` isn't currently present).  It is also not currently iterable and there are likely other compatibility issues I'll need to address.  Or we could not try :-)

ladyada, 2018-12-27:
we use the `neopixel` library a lot, so if we can at least have a transitionary period where it uses either pixelbuf or the built in pixel manager, with a try/import/except attempt, that would be best. 
lets chat with @tannewt when he's back since there's a lot of API changes and it looks like he was part of the discussion. maybe next thursday Jan 3rd meeting if eveyone's around :)

tannewt, 2019-01-03:
I think we should aim to be backwards compatible with `neopixel` and break any weird things with other RGB LED drivers. Most of those differences are a result of bad copy and paste so we should take the hit now to make them uniform.

We should update each individual driver and include the new PixelBuf dependency as needed.

Does that clarify things? Any other issues we need direction on?

kevinjwalters, 2019-03-21:
Are there any examples on how to use PixelBuf https://circuitpython.readthedocs.io/en/latest/shared-bindings/_pixelbuf/__init__.html

How does this compare memory-wise with the NeoPixel/DotStar libraries for CircuitPython 3.x?

rhooper, 2019-03-21:
At some point I hope my life becomes less busy, otherwise I'll work on this during sprints at PyCon.
https://github.com/rhooper/CircuitPython_Pixel_Driver is the WIP repo that has neopixel.py and adafruit_dotstar.py drivers.  Memory-wise it should be similar to the current implementations, unless you use the single buffered mode that doesn't refresh the brightness of all pixels when brightness is changed.  That functionality isn't exposed in the backwards compatible APIs in the above linked repo.

Hit me up on Discord with questions if you like.

FutureSharks, 2019-04-01:
Any update on this?

I'm trying to write to a ring of 16 pixels to show the position of an encoder when it is turned by a human. Currently the lag makes it feel quite unresponsive on my Itsy Bitsy M0:

```
num_pixels = 16
 
pixels = neopixel.NeoPixel(pixel_pin, num_pixels, brightness=0.1, auto_write=True, pixel_order=(1, 0, 2, 3))

for i in range(num_pixels):
    pixels[i] = (encoder_position, 0, 0, 0)
```

Currently it takes about 250ms to fill the whole ring.

ladyada, 2019-04-01:
shark, turn off auto_write, also use `fill()`

dhalbert, 2019-04-01:
@FutureSharks Besides the `_pixelbuf` work described above, in progress:

1. Set your `brightness=1.0` (or omit the arg: it defaults to 1.0) which will speed things up, and make the color values smaller to compensate. When `brightness` is not 1.0 the pixel values must be copied before sending out the data.

2. Set `auto_write=false`. Set the pixel values in that `for` loop, and then after the loop, do `pixels.show()`. That will suppress sending any data until it's all ready to send.

kevinjwalters, 2019-04-01:
Your question is probably better placed in the forums.

Try `auto_write=False` and after the `for` loop do a `pixels.show()`. What does the 4 tuple do, I normally see those as (red, green, blue) values? Also if you want it even quicker use brightness=1.0 and control brightness by reducing the values you are setting.

FutureSharks, 2019-04-01:
> shark, turn off auto_write, also use fill()

Thanks @ladyada but I don't want to fill the whole ring with the same colour. The ring is supposed to show the position of the encoder to a human so if the ring is all one colour it doesn't show anything.

I tried disabling `auto_write` but performance is the same:

```
while True:
    for i in range(num_pixels):
        pixels[i] = (100, 0, 0, 0)
        pixels.show()
    
    for i in range(num_pixels):
        pixels[i] = (0, 0, 100, 0)
        pixels.show()
    
    count += 1
    
    if count % 10 == 0:
        print('It took {0} seconds to write to ring'.format((time.time() - start) / 10 / 2))
        start = time.time()
```

> Besides the _pixelbuf work described above

I get a memory allocation error `MemoryError: memory allocation failed, allocating %u bytes
` when testing [CircuitPython_Pixel_Driver](https://github.com/rhooper/CircuitPython_Pixel_Driver) code.

> Set your brightness=1.0 (or omit the arg: it defaults to 1.0) which will speed things up

Thanks @dhalbert this helped a lot! The difference is `0.3` seconds vs `0.05` seconds.

Thanks again for the replies, I think I'm good now 💙

dhalbert, 2019-04-01:
Move the `.show()` out of the loop:
```for i in range(num_pixels):
        pixels[i] = (100, 0, 0, 0)
    pixels.show()
```

PTS93, 2019-12-25:
What is the current state of this?
I would love to do some animations with a 120px Neopixel strip but the refresh rate is abysmal with the default library :/
I could only find the api doc but not really sure on how this is supposed to be used, does it tie into the standard neopixel library?

ladyada, 2019-12-30:
hiya we're still hacking away at the PR - please help test when its released!

PTS93, 2019-12-30:
Gladly! Lemme know when there is something to test.

jc78, 2020-01-01:
I've run into this same problem, slow animations. I'd also be happy to test this once its released.

kevinjwalters, 2020-01-04:
FYI, there's also an issue ticket for `fill()` being slow over on https://github.com/adafruit/Adafruit_CircuitPython_DotStar/issues/37

PTS93, 2020-01-04:
As I didn't see it discussed in this issue, is there any plans to bring over the DMA based libaries to cpy?

rhooper, 2020-01-04:
@PTS93 the latest changes to _pixelbuf were just merged into master yesterday.  If you're good with building CircuitPython yourself, you can try the _pixelbuf variants of neopixel.py and adafruit_dotstar.py over at https://github.com/rhooper/Adafruit_CircuitPython_DotStar/tree/pixelbuf and https://github.com/rhooper/Adafruit_CircuitPython_NeoPixel/tree/pixelbuf

If you're using one of the new nRF boards like the Feather nRF52840 https://www.adafruit.com/product/4062 DMA is used for neopixel_write (if possible).

There's probably no reason that the M4 boards couldn't use the techniques the NRF neopixel_write and https://github.com/adafruit/Adafruit_NeoPXL8/blob/master/Adafruit_NeoPXL8.cpp use to do DMA (PWM) for more boards.  I'm guessing there's likely not enough space/memory to do the DMA technique for M0.

rhooper, 2020-01-04:
@kevinjwalters _pixelbuf has a native fill that the previously mentioned forks/branches of adafruit_dotstar.py and neopixel.py can use.

PTS93, 2020-01-04:
Thanks I will give it a try! Building it is no problem.
Though regarding at DMA for M0 needing to much space/memory, I don't really understand, thats the main target of the Arduino libraries?

PTS93, 2020-01-04:
So I got the latest master and compiled for the PyBadge but speed is exactly the same, is there anything different about how I need to call or construct anything?
import neopixel
```neopixels = neopixel.NeoPixel(board.D3, 120, brightness=0.3, auto_write=False, pixel_order=neopixel.GRB)```

Using the hello world code of the py badge:
``` python
        for color in range(0, 360, speed):
	    for index in range(0, NEOPIXEL_COUNT):
	        palette_index = pixels[index] + color * direction
	        if palette_index >= 360:
	            palette_index -= 360
	        elif palette_index < 0:
	            palette_index += 360
	        neopixels[index] = palette[palette_index]
	    neopixels.show()
```
This animation takes about 2 seconds to complete. `neopixels[index] = palette[palette_index]` seems to be the biggest time hogger which I assume pixbuf is trying to address?

PTS93, 2020-01-04:
Ok I compiled the pixbuf tree and it works a lot faster now!
https://twitter.com/timonsku/status/1213599823116591105

Still a lot slower than the Arduino library but it now makes 100-200 pixel strips usable in combination with other stuff, which before was just impossible! Thanks for all the work!

sta-c0000, 2020-02-27:
😕 I'm surprised 5.0.0 stable release appears might _not_ include new _pixelbuf driven NeoPixel?!  Expected that would be one of its main features.

For those who want to try out _pixelbuf accelerated neopixels with 5.0.0 without building / compiling CircuitPython...

You can easily:
- Download [neopixel.py](https://raw.githubusercontent.com/rhooper/Adafruit_CircuitPython_NeoPixel/pixelbuf/neopixel.py) from the pixelbuf branch.
- Rename `neopixel.py` to `pixelbuf_neopixel.py` (_or anything different_)
- copy it to `/lib` folder (create if needed) on your circuitpython board
- to use: `import pixelbuf_neopixel as neopixel`

Simple example:
```python
import board
import pixelbuf_neopixel as neopixel
# import neopixel # uncomment to compare speed
pixels = neopixel.NeoPixel(board.NEOPIXEL, 10) # pin, # of neopixels

col = [127,255,64]

max_range = max(col)

while True:
    for i in range(max_range + 1):
        # gamma 2.0 adjust
        c = [max(0, n-i)**2//255 for n in col]
        pixels.fill(c)
    for i in range(max_range, 0, -1):
        # gamma 2.0 adjust
        c = [max(0, n-i)**2//255 for n in col]
        pixels.fill(c)
```

tannewt, 2020-03-03:
@sta-c0000 sit tight. We can re-release neopixel now that 5.0.0 is stable. We couldn't until then because the driver hard crashes on 4.1.2.

wallarug, 2020-03-05:
How does this actually work behind the scenes to create a faster library?

tannewt, 2020-03-05:
PixelBuf speeds up RGB pixels by using C to do math on all the pixels such as the pre-transmit application of brightness. The libraries need to inherit from PixelBuf in order to get this speed-up though.

paulmand3l, 2020-03-06:
Looks like the adafruit_pypixelbuf.mpy available in the 5.x.x library downalod from circuitpython.org still requires `re`:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 4, in <module>
  File "/lib/pixelbuf_neopixel.py", line 38, in <module>
  File "adafruit_pypixelbuf.py", line 32, in <module>
ImportError: no module named 're'
```

dhalbert, 2020-03-06:
@paulmand3l You're right, we forgot to do a release after fixing the `re` issue. I jus made https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf/releases/tag/1.0.3, which has the `re` fix.

paulmand3l, 2020-03-21:
Hmm, now I'm getting

```
Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 12, in <module>
  File "/lib/neopixel.py", line 132, in __init__
TypeError: function missing required positional argument #2
```

paulmand3l, 2020-03-21:
Ah, nm. Realized I wasn't running 5.x circuitpython. Upgraded to 5.x and it works now. Boosted my frame rate from 4 to 30fps with 24 pixels. Thanks!

tannewt, 2020-07-13:
Ok, PixelBuf has been released and everyone loves the speed. Huge thanks to @rhooper for PixelBuf and @dunkmann00 for helping get it finished.

closed 2020-07-13 by tannewt