adafruit/circuitpython #886: audioio allocates extra 512 bytes of ram on play in CP 3.0

deshipu, 2018-05-30:
Since audioio has been reworked in 3.0, it now allocates two buffers of 512 bytes each when `.play()` is called on a file: https://github.com/adafruit/circuitpython/blob/master/shared-module/audioio/WaveFile.c#L102-L112

That is problematic for large programs that already use most of memory, because they can fail pretty much randomly at any moment due to memory fragmentation.

In 2.x, the buffers were allocated once, when the `AudioInOut` object was created:
https://github.com/adafruit/circuitpython/blob/2.x/atmel-samd/common-hal/audioio/AudioOut.c#L363-L375

What I would like to add is the ability to pass pre-allocated buffers (possibly even smaller than 512 bytes) that would be used for this — this way I can create them in advance at the beginning of the program, avoiding memory fragmentation, re-use them for playing different sounds, and even share with other functions that need buffers, such as SPI communication with the display (as long as they don't need them at the same time). Those parameters would be optional, and by default the buffers would be created as they are now.

deshipu, 2018-05-30:
Hmm, actually it's allocated when `WaveFile` is created, I thought it was in `.play()` because I create it just before calling `.play()`. I think the difference from the old code is that previously the buffer was GC-ed when I deleted the old `AudioInOut` object and created a new one to play a new sound, but with `WaveFile`, the old object somehow stays around, and i get a `MemoryError` when I try to create a new one.

deshipu, 2018-05-30:
In any case, the possibility of sharing the buffers between different `WaveFile` objects would be helpful.

tannewt, 2018-05-31:
I think the audio stuff will get some attention along with the upcoming gaming handheld in the long term.

In the short term, could you try clearing `second_buffer` here as well? https://github.com/adafruit/circuitpython/blob/master/shared-module/audioio/WaveFile.c#L116 No need to keep it around if the object is deinited but still in a namespace.

deshipu, 2018-09-30:
Clearing the second buffer made no difference.

Ultimately I didn't manage to get this to work. I ended up freezing some of the assets to make room in memory for the audio buffers. I decided that I need to release the 3.x firmware for µGame, since 4.x is already on the horizon, and I can't wait any longer.

kevinjwalters, 2020-06-15:
Is this ticket still relevant?

tannewt, 2020-07-14:
@deshipu Is this still an issue with 6.x?

deshipu, 2021-04-02:
Fixed by #2032

closed 2021-04-02 by deshipu

ladyada, 2021-04-02:
yayy! :)