adafruit/circuitpython #897: Cannot record and play back audio at the same time

kattni, 2018-06-02:
Discovered by attempting to add a `sound_level` functionality to the CPX `Express` class library. Checking the sound level requires recording to a small buffer, and If used in code with anything that plays audio (like `play_tone`), it caused the filesystem to hang and required a hard reset to regain functionality. It would at times run for a few seconds before failing, but otherwise it failed every time. I was able to get the two functions to exist together in code if I had `sound_level` check to see if there was anything being played back and have it `pass` if something was being played - meaning the two never actually worked together.

These issues were present in 2.x and 3.0.

It was identified as an issue with audioio.

dhalbert, 2018-06-03:
Could you try this with the absolute latest 3.0 code? @tannewt fixed some stuff with interrupts in #887, and I can no longer duplicate the PDMIn problem in #879 after I update to the tip of master (which is a good thing!).

kattni, 2018-06-05:
I am able to use `sound_level`, `loud_sound`, `start/stop_tone` and `play_tone` in the same code with the latest build of 3.0. `sound_level` stops printing while it's playing something, but it doesn't crash and continues to work.

The `sound_level` code eventually hangs. The hanging has been identified as potentially the same issue as in #879. Closing this issue in favor of that one.

closed 2018-06-05 by kattni