adafruit/circuitpython #899: Gaming API!

tannewt, 2018-06-05:
Improve our existing APIs for game creation and add more as needed.

Please dump links and ideas here!

deshipu, 2018-06-05:
There is a very nice fantasy console with a big community called pico8 (https://www.lexaloffle.com/pico-8.php), but it's closed-source and LUA. Then again, there is a similar open source project with Python support written in Rust: https://github.com/Gigoteur/UnicornConsole

The löve2d engine (https://love2d.org/) is also LUA, but they are very popular and have good API.

deshipu, 2018-06-05:
I also need to add a 2x2 pixel mode to the Stage library (https://github.com/python-ugame/circuitpython-stage) so that it can be used with those huge 320x240 displays without having to strain your eyesight to see the 16x16 sprites.

deshipu, 2018-06-05:
If you want to see some examples of PyGame games, there is a bunch that I wrote over the years: http://sheep.art.pl/Games

tannewt, 2018-06-06:
Looks like most of the fantasy consoles have single files they use to distribute. Should we have ours be zip files that can live zipped on the fs or require they be unzipped into a folder?

ladyada, 2018-06-06:
personally - i think its ok to tell people they need to put raw files down in a folder, that way we dont have to write to the FS which can cause corruption

deshipu, 2018-06-18:
Some more fantasy console links: https://alexschroeder.ch/wiki/2018-05-15_Game_Development

tannewt, 2018-06-19:
Apple's SpriteKit: https://developer.apple.com/documentation/spriteKit

kattni, 2018-07-06:
Someone suggested looking at TIC-80 as well. It's open source: https://tic.computer/

superkahuna, 2018-07-12:
I'm not sure what APIs you have already, but having basic physics, movement, basic collision detection and easing functions would go a long way to adoption. I used a lot of game engines and honestly those functions are not hard to write from scratch, at least the basics. I'm not talking about box2d physics, pixel perfect collision or anything that would over tax the MCU, just the basics.

I've personally found this site helpful (note, the code is processing not python but is easy to understand):  https://natureofcode.com/book/introduction/

tannewt, 2018-07-12:
@superkahuna I started with https://github.com/python-ugame/circuitpython-stage in my testing. I ported Celeste from PICO-8. My plan is to add C apis where we need the performance.

tannewt, 2018-08-13:
PICO-8 talk worth watching: https://www.youtube.com/watch?v=87jfTIWosBw

deshipu, 2018-09-07:
This is a very nice video that discusses some techniques for saving memory: https://www.youtube.com/watch?v=ZWQ0591PAxM

davidperrenoud, 2018-09-16:
You can take example from the Arduboy game format:
https://community.arduboy.com/t/arduboy-game-format-guide-arduboy-files/2784

It is a zip file ending in _.arduboy_ which contains a precompiled binary and some metadata:
* game.hex
* info.json
* screenshot00.png
* banner.png

For CircuitPython, this could be as simple as shipping a .uf2 file with one or more games inside it. The advantage of this solution instead of shipping Python files, is that you can evolve the gaming API along with the games.

If you ship a .uf2 gaming library separately to the games, you run the risk of having a version mismatch and compatibility problems in the long run. This is what happened with the _Arduboy_ vs. _Arduboy2_ libraries. Also, some developers may like to optimise your gaming API in ways you wouldn't have anticipated.

davidperrenoud, 2018-09-17:
This is also how [GP Blocks](https://gpblocks.org/faq/) and [MicroBlocks](http://microblocks.fun) works:
* .gpp file (a zip file)
  * objects.gpod
  * thumbnail.png
  * ancestor
  * hash
  * topHash
  * modules/
    * 6b3f57d31cf79cb743a589a0f3862f09493fe246ab02a8b9bf44132136f14f38

The code, sounds and images are all saved in the .gpod file which I assume stands for _GP Object Database_. You could also save the code as separate .gp files, but then you lose some of the advantages of a [Smalltalk-like image system](https://en.wikipedia.org/wiki/Smalltalk#Image-based_persistence).

greg76, 2019-01-15:
Pythonista for iOS has a very nice little 2D Scene package made for making simple games. I think the API design is easy to pick up and effective. http://omz-software.com/pythonista/docs/ios/scene.html
I like the event handling and how animations can be tied to scene graph elements a lot.

tannewt, 2019-08-23:
With the display speedups I think we can close this. There is certainly more work to do for audio and potentially collisions but we should have specific issues for those as needed.

closed 2019-08-23 by tannewt