adafruit/circuitpython #900: samd21/51: Hang issue when running tests on the board

notro, 2018-06-05:
I'm unable to reproduce the problem in #894, but I was able to get another one.

I order to get the tests running at all I first had to do a:
```python
storage.erase_filesystem()
```
Otherwise I got the same hung error mentioned further down.

I've run the tests 3 times in a row, no problem. 
```
.../tests $ CPBOARD_EXEC_MODE=disk python3 run-tests --device=metro_m4_express --target=pyboard -d basics circuitpython
<snip>

413 tests performed (12423 individual testcases)
401 tests passed
24 tests skipped: array_micropython builtin_compile builtin_delattr builtin_pow3 builtin_pow3_intbig class_delattr_setattr class_descriptor class_new class_notimpl class_reverse_op class_store_class class_super_object exception_chain fun_name namedtuple1_cpython_compat nvm_not_present nvm_present object_dict object_new ordereddict1 ordereddict_eq parser struct_micropython subclass_classmethod
12 tests failed: async_await async_await2 async_def async_for async_for2 async_with async_with2 blinky enumerate errno1 struct1 struct2

```
Then I press the board reset button and restart run-tests and it just hangs.
After a while I get this in the log:
```
[ 2947.012329] INFO: task scsi_eh_0:69 blocked for more than 120 seconds.
[ 2947.012343]       Tainted: G         C      4.14.34-v7+ #1110
[ 2947.012348] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2947.012356] scsi_eh_0       D    0    69      2 0x00000000
[ 2947.012398] [<8079a0b8>] (__schedule) from [<8079a730>] (schedule+0x50/0xa8)
[ 2947.012415] [<8079a730>] (schedule) from [<8079aba8>] (schedule_preempt_disabled+0x18/0x1c)
[ 2947.012435] [<8079aba8>] (schedule_preempt_disabled) from [<8079c4b0>] (__mutex_lock.constprop.3+0x190/0x58c)
[ 2947.012453] [<8079c4b0>] (__mutex_lock.constprop.3) from [<8079c9c8>] (__mutex_lock_slowpath+0x1c/0x20)
[ 2947.012469] [<8079c9c8>] (__mutex_lock_slowpath) from [<8079ca28>] (mutex_lock+0x5c/0x60)
[ 2947.012487] [<8079ca28>] (mutex_lock) from [<805ea52c>] (device_reset+0x28/0x54)
[ 2947.012505] [<805ea52c>] (device_reset) from [<80571424>] (scsi_try_bus_device_reset+0x34/0x58)
[ 2947.012521] [<80571424>] (scsi_try_bus_device_reset) from [<80572be8>] (scsi_eh_ready_devs+0x25c/0x88c)
[ 2947.012536] [<80572be8>] (scsi_eh_ready_devs) from [<805741dc>] (scsi_error_handler+0x3a0/0x3a4)
[ 2947.012553] [<805741dc>] (scsi_error_handler) from [<8013d860>] (kthread+0x13c/0x16c)
[ 2947.012572] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[ 2947.012584] INFO: task usb-storage:71 blocked for more than 120 seconds.
[ 2947.012590]       Tainted: G         C      4.14.34-v7+ #1110
[ 2947.012595] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2947.012600] usb-storage     D    0    71      2 0x00000000
[ 2947.012622] [<8079a0b8>] (__schedule) from [<8079a730>] (schedule+0x50/0xa8)
[ 2947.012637] [<8079a730>] (schedule) from [<8079e314>] (schedule_timeout+0x1d0/0x3f4)
[ 2947.012653] [<8079e314>] (schedule_timeout) from [<8079b3b4>] (wait_for_common+0xc0/0x184)
[ 2947.012669] [<8079b3b4>] (wait_for_common) from [<8079b498>] (wait_for_completion+0x20/0x24)
[ 2947.012685] [<8079b498>] (wait_for_completion) from [<805b3314>] (usb_sg_wait+0x174/0x188)
[ 2947.012704] [<805b3314>] (usb_sg_wait) from [<805eaffc>] (usb_stor_bulk_transfer_sglist.part.2+0x88/0xec)
[ 2947.012723] [<805eaffc>] (usb_stor_bulk_transfer_sglist.part.2) from [<805eb0bc>] (usb_stor_bulk_srb+0x5c/0x64)
[ 2947.012740] [<805eb0bc>] (usb_stor_bulk_srb) from [<805eb1e8>] (usb_stor_Bulk_transport+0x124/0x370)
[ 2947.012758] [<805eb1e8>] (usb_stor_Bulk_transport) from [<805ebaac>] (usb_stor_invoke_transport+0x30/0x4bc)
[ 2947.012776] [<805ebaac>] (usb_stor_invoke_transport) from [<805ea7b0>] (usb_stor_transparent_scsi_command+0x18/0x1c)
[ 2947.012794] [<805ea7b0>] (usb_stor_transparent_scsi_command) from [<805ecf7c>] (usb_stor_control_thread+0x198/0x2a0)
[ 2947.012811] [<805ecf7c>] (usb_stor_control_thread) from [<8013d860>] (kthread+0x13c/0x16c)
[ 2947.012827] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[ 2947.012931] INFO: task python3:1658 blocked for more than 120 seconds.
[ 2947.012938]       Tainted: G         C      4.14.34-v7+ #1110
[ 2947.012942] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2947.012947] python3         D    0  1658   1657 0x00000000
[ 2947.012973] [<8079a0b8>] (__schedule) from [<8079a730>] (schedule+0x50/0xa8)
[ 2947.012988] [<8079a730>] (schedule) from [<8014b0b4>] (io_schedule+0x20/0x40)
[ 2947.013004] [<8014b0b4>] (io_schedule) from [<8021c9b4>] (wait_on_page_bit+0x110/0x130)
[ 2947.013019] [<8021c9b4>] (wait_on_page_bit) from [<8021cab4>] (__filemap_fdatawait_range+0xe0/0x114)
[ 2947.013033] [<8021cab4>] (__filemap_fdatawait_range) from [<8021cb54>] (filemap_fdatawait_keep_errors+0x34/0x60)
[ 2947.013050] [<8021cb54>] (filemap_fdatawait_keep_errors) from [<802ba9ec>] (sync_inodes_sb+0x20c/0x24c)
[ 2947.013068] [<802ba9ec>] (sync_inodes_sb) from [<802c02c0>] (sync_inodes_one_sb+0x24/0x28)
[ 2947.013088] [<802c02c0>] (sync_inodes_one_sb) from [<8028ddec>] (iterate_supers+0xf0/0x150)
[ 2947.013106] [<8028ddec>] (iterate_supers) from [<802c0664>] (sys_sync+0x44/0xb4)
[ 2947.013122] [<802c0664>] (sys_sync) from [<80108060>] (ret_fast_syscall+0x0/0x28)
```
Unplugging the board gives:
[... 37 lines cut ...]
[ 5072.578317] input: Adafruit Industries LLC Metro M4 Express as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.3/0003:239A:8021.0004/input/input3
[ 5072.645921] hid-generic 0003:239A:8021.0004: input,hidraw0: USB HID v1.11 Keyboard [Adafruit Industries LLC Metro M4 Express] on usb-3f980000.usb-1.4/input3
[ 5073.615766] scsi 0:0:0:0: Direct-Access                                    PQ: 0 ANSI: 3
[ 5073.619857] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 5073.621912] scsi 0:0:0:1: Direct-Access                                    PQ: 0 ANSI: 3
[ 5073.622787] sd 0:0:0:1: Attached scsi generic sg1 type 0
[ 5073.623302] sd 0:0:0:0: [sda] 4089 512-byte logical blocks: (2.09 MB/2.00 MiB)
[ 5073.625166] sd 0:0:0:0: [sda] Write Protect is off
[ 5073.625184] sd 0:0:0:0: [sda] Mode Sense: 00 00 00 00
[ 5073.626508] sd 0:0:0:0: [sda] Asking for cache data failed
[ 5073.626523] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 5073.627389] sd 0:0:0:1: [sdb] Attached SCSI removable disk
[ 5073.644089]  sda: sda1
[ 5073.647533] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 5074.751345] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

```

Now I can run the tests again.

I've had many different hang situations when working on #893, but I haven't paid attention to the details. I first press reset, if that doesn't work I unplug the board, if that doesn't work I upload the firmware anew (not sure if this has ever helped), and if that doesn't work I use the flash eraser.
Most of the testing for #893 was done on a Feather M0 Express, and some on a Metro M4 Express.

I can't spend more time working on this, but I've made a bug report after request by @tannewt.
Just close this if it can't be reproduced.

tannewt, 2018-06-05:
Thanks for the issue! Maybe there is state that isn't reset across button presses. I'll look more into it.

sommersoft, 2018-06-05:
My first guess with the above `test -> reset button -> improper unmount` is that one of the tests is running an operation on the FS, and when the board isn't properly unmounted from the host that is gumming up the reset. Which is a documented problem, of course. 

I don't have any experience with the tests, so I may be off on that theory.

jepler, 2018-06-06:
I found that on my metro m4 I could reproduce a weird hang with the following interactively, freeing me from repeatedly reinitializing the SPI flash or other long debugging cycles:
```
>>> exec("def f(): yield from f()\ntry: list(f())\nexcept: raise")
```

When I run this, serial terminal hangs but I can break in gdb:
```
Program received signal SIGINT, Interrupt.
HardFault_Handler () at supervisor/port.c:295
295         asm("");
(gdb) where
#0  HardFault_Handler () at supervisor/port.c:295
#1  <signal handler called>
#2  gc_collect_root (ptrs=<optimized out>, len=53) at ../../py/gc.c:334
#3  0x01000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) frame 2
#2  gc_collect_root (ptrs=<optimized out>, len=53) at ../../py/gc.c:334
334         gc_drain_stack();
(gdb) p $sp
$5 = (void *) 0x20021790 <mp_state_ctx+432>
(gdb) p/x mp_state_ctx.thread.stack_top - mp_state_ctx.thread.stack_limit
$6 = 0x200219b8
(gdb) p mp_state_ctx.thread.stack_top - $sp
$7 = 58880
(gdb) p mp_state_ctx.thread.stack_limit
$8 = 58328
```

.. so we are somewhat "below" where the stack is supposed to be able
to reach but we should be in the 1kB safe region...

but wait.... !  It turns out the stack pointer has invaded the BSS segment:

```
(gdb) p &_ezero
$12 = (<data variable, no debug info> *) 0x20021820
```
restarting,

```
Breakpoint 1, main () at ../../main.c:244
244         mp_stack_ctrl_init();
(gdb) p $sp
$13 = (void *) 0x2002fd68
(gdb) p estack
No symbol "estack" in current context.
(gdb) p &_estack
$14 = (<text variable, no debug info> *) 0x2002fff8
(gdb) p 0x2002fff8 - 0x2002fd68
$15 = 656
```

.. so more than half the 1024-byte safety factor has been eaten up by stack above `main()` (or maybe above `mp_stack_ctrl_init()`, leaving an inadquate safety factor for running gc right at the moment when the stack has already been blown.

Instead of setting the stack bounds with `mp_stack_ctrl_init` which tries to guess the top of stack from the current stack, we can instead initialize it with `mp_stack_set_top` and an explicit address.  PR to follow.

PR #903 (2018-06-06): main: Explicitly set the stack top based on _estack
.. setting it based on the ad-hoc stack pointer calculation of `mp_stack_ctrl_init()` meant that the stack used above `main()` counts against the 1KiB safety factor that the `mp_stack_set_limit` call tries to establish.  It turns out, at least on M4, that over half of the safety factor is used up by stack-above-main()!

In the case of the `basics/gen_stack_overflow.py` test, which blows the stack on purpose, it turns out that gc would be called while handling the "maximum recursion depth exceeded" error, and this needed more stack than was le [... cut]
commit a4e06bd9 (2018-06-06): main: Explicitly set the stack top based on _estack

.. setting it based on the ad-hoc stack pointer calculation of
mp_stack_ctrl_init() meant that the stack used above main() counts
against the 1KiB safety factor that the mp_stack_set_limit call tries
to establish.  It turns out, at least on M4, that over half of the
safety factor is used up by stack-above-main()!

In the case of the basics/gen_stack_overflow.py test,
which blows the stack on purpose, it turns out that gc would be called
while handling the "maximum recursion depth exceeded" error, and this
needed more stack than was left.

Cl [... cut]
closed 2018-06-06 by tannewt