adafruit/circuitpython #912: Getting files onto NRF52

No1089, 2018-06-07:
Hi,

I finally received my J-link and programmed three different NRF52 boards, but on the one I want to use it most (Ruuvi tag), I am not sure how to load the code. It does not have a serial connection that is enabled(and I have no idea how to get it running), nor a USB port. The WebREPL via webBluetooth is working, but none of the REPL file transfer programs (or Ampy) supports that. I do have a Bluefruit LE Friend (UART) but for that I first need the NRF52 to connect etc.

Is it possible to use SWD to upload the code, instead of transferring the files?

deshipu, 2018-06-07:
I think the WebREPL had a file upload functionality?

No1089, 2018-06-07:
@deshipu I'm an idiot. I was using the Espruino webREPL - just googled micropython WebREPL - and yes it has that functionality. The documentation on the NRF52 could use some editing - a few steps were missing/unexplained. Thanks.

No1089, 2018-06-07:
https://micropython.org/webrepl/? does not allow me to use WebBluetooth :(
This is what I was using: https://glennrub.github.io/webbluetooth/micropython/repl/

jerryneedell, 2018-06-07:
There are instructions for adding support for the WebBluetooth here. 
https://github.com/adafruit/circuitpython/tree/master/ports/nrf
It is not enabled by default so you will have to create a custom build.

No1089, 2018-06-07:
I have done so, and I can access the REPL with this: (https://glennrub.github.io/webbluetooth/micropython/repl/), but unlike (https://micropython.org/webrepl/), there is no option to upload a file so I can get my code running.

jerryneedell, 2018-06-07:
Ah - sorry -- I just reread your earlier posts and see that. 
Its's not pretty, but I suppose you could  compile your code in as a "frozen" module.

No1089, 2018-06-07:
@jerryneedell Apologies for my ignorance - but how do I create a frozen module?

jerryneedell, 2018-06-07:
It's been awhile since I played with this, but there is an example here:
https://github.com/adafruit/circuitpython/tree/master/ports/nrf/freeze
This is compiled into your build. I think any .py file in the "freeze" folder will also get compiled in. You can then invoke it from the REPL.   I don't recall if it gets stored as "freeze/test" or as test.
If you look at '''
help('modules') '''
should should see it. I'll try to provide an example tonight or tomorrow if you have not gotten it working first.

jerryneedell, 2018-06-07:
see: https://github.com/adafruit/circuitpython/blob/master/ports/nrf/Makefile#L360
for how the Mkefile handles it.
in line 36 ```FROZEN_MPY_DIR = freeze```

jerryneedell, 2018-06-07:
Here is an example executing the freeze/test.py  build into the current master
```
Adafruit CircuitPython 3.0.0-beta.0-43-gde61bd0 on 2018-06-02; Bluefruit nRF52 Feather with NRF52832
>>> 
>>> 
>>> import test
>>> test.hello()
Hello nrf52!
>>> 
```

jerryneedell, 2018-06-07:
better  example- I copied thhese files to the freeze folder"
```
erryneedell@Ubuntu-Macmini:~/projects/adafruit_github/circuitpython_master/ports/nrf$ cat freeze/blinky.py 
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.LED2)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/circuitpython_master/ports/nrf$ cat freeze/ble_scan.py 

from ubluepy import Scanner, constants

def display_scan_results(scan_entries):
    for e in scan_entries:
        print("ADDR:  ", e.addr())
        print("TYPE:  ", e.addr_type())
        print("RSSI:  ", e.rssi())

        # Parse the contents of the advertising packet
        scan = e.getScanData()
        if scan:
            for s in scan:
                # Convert byte array to hex format string
                hex = ' '.join('0x%02X' % b for b in s[2])
                # Display enum value and hex string together
                print('\t{}: {}'.format(s[1], hex))

        # Line break between record sets
        print("")

# Scan 1s for advertising devices in range
s = Scanner()
scan_res = s.scan(1000)

# Display the scan results
display_scan_results(scan_res)
```

rebuilt and uploaded then executed them:

```
Adafruit CircuitPython 3.0.0-beta.0-60-g74ced17 on 2018-06-07; Bluefruit nRF52 Feather with NRF528
32
>>> import blinky
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "blinky.py", line 10, in <module>
KeyboardInterrupt: 
>>> import ble_scan
ADDR:   c8:69:cd:1c:5d:f8
TYPE:   0
RSSI:   -84
        AD_TYPE_FLAGS: 0x1A
        AD_TYPE_MANUFACTURER_SPECIFIC_DATA: 0x4C 0x00 0x10 0x05 0x01 0x10 0xC8 0x70 0xC0

ADDR:   c8:69:cd:1c:5d:f8
TYPE:   0
RSSI:   -84

lots more followed....
```

jerryneedell, 2018-06-07:
building code tnit "freze" is certainly not ideal for testing but it does work if you have no other way to get code uploaded. It would be great to have  upload capability added to the BLE REPL

Good luck!

No1089, 2018-06-08:
@jerryneedell Fantastic! Thank you. And thank you for the examples. I just had a chance to test and it is working perfectly! Now at least I can get going, I don't mind that it takes a while currently.

deshipu, 2018-06-08:
This is a bit hacky, but you could create a `main.py` that imports it with:
```python
with open("main.py", "w") as f:
    f.write("import yourfile")
```

jerryneedell, 2018-06-08:
but how can you get the main.py file loaded :-(
I tried just naming the frozen file main.py, but that did not work. It runs if I "import main"

jerryneedell, 2018-06-08:
never mind - I see what you did there ;-)

jerryneedell, 2018-06-08:
@deshipu suggestion works great!

jerryneedell, 2018-06-08:
then to remove it" control-c and enter REPL ```i
import os
os.remove("main.py")
```
nice!

No1089, 2018-06-08:
Really sorry for my ignorance here, but @deshipu - I created a main.py with the code you posted and I am just getting an error when I try to run it. I'm assuming I should be doing something different. Could you perhaps post an example as you did previously @jerryneedell ? I typed the command in directly, but no luck.

jerryneedell, 2018-06-08:
Assuming you have blinky.py in the freeze folder and have loaded the version of CP with it so you can run ```
import  blinky
``` 
ok then open REPL and at the prompt type.
```
with open("main.py", "w") as f:
    f.write("import blinky")
```
then reboot and it should run on boot. Demo in a few minutes.

jerryneedell, 2018-06-08:
here is a demo:
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.0.0-beta.0-75-gc01ce17 on 2018-06-08; Bluefruit nRF52 Feather with NRF52832
>>>
>>> with open("main.py","w") as f:
...     f.write("import blinky")
...
...
...
13
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:

```
LED is blinking

No1089, 2018-06-08:
When the ... appears, do I just wait or press enter?
This is what I am seeing:
`>>> with open("main.py", "w") as f:
...     f.write("import blinky")
...     
...     
... 
Traceback (most recent call last):
  File "<stdin>", line 2, in <module> OSError: 5 >>>`

jerryneedell, 2018-06-08:
hit enter 3 times to execute the write -- looks like you did that and got OSError 5 -- that is an I/O error.
No idea why.

No1089, 2018-06-08:
Tried it with a different board just now - Feather52 instead of the RuuviTag - same problem (OSError: 5). Heading home and will check again. Thanks for the help so far.

jerryneedell, 2018-06-08:
FYI - I am using a feather52 as well.  Good luck!

jerryneedell, 2018-06-08:
One difference - I am not using the BLE REPL  -- using USB for the feather52.  Not sure if or why it may matter.

deshipu, 2018-06-08:
Well, you put it on the filesystem.

jerryneedell, 2018-06-08:
I just confirmed that when using BLE REPL - attempting to write fails with OSError 5

jerryneedell, 2018-06-08:
It looks like there is some difference in the FS initialization when BLE REPL is enabled. I noticed that at some point in my testing, recreated the FS leaving only a lib folder.  This done in supervisor/filesystem.h if no FS is found.  SO wahtever is going on , when BE REPL is enabled, the FS appears to be read-only to python while via the USB REPL it is writeable. As I recall, this was a bit of a mystery anyway. Note
in the nrf build 
CIRCUITPY_BOOT_OUTPUT_FILE is not defined.
```
//#define CIRCUITPY_BOOT_OUTPUT_FILE "/boot_out.txt"

```
the is a lot going on with the FS at startup. Still trying to understand what is different under BLE REPL

tannewt, 2018-12-07:
Closing this in favor of #1010

closed 2018-12-07 by tannewt