adafruit/circuitpython #924: prettify Adafruit_Pixie library

ladyada, 2018-06-12:
we have an arduino library for these mindblasting pixels
https://www.adafruit.com/product/2741
i just made a short example that mimics NeoPixel's format but it needs a repo, splitting up into a proper module etc!
https://forums.adafruit.com/viewtopic.php?f=52&t=136796&p=677580#p677580
no rush on this one
see the circuitpy neopixel library for inspiration
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L78

if you can get it to cookiecutter+travis point, ill do another set of testing since i have a strand of 3 pixies for testing porpoises

tannewt, 2018-06-12:
@rhooper may be interested to see this as he has been thinking about reorganizing the code underneath the rgb led libraries.

kattni, 2018-07-18:
This is finished! Lib is set, guide is updated and both have been blogged!

closed 2018-07-18 by kattni