adafruit/circuitpython #929: RTC on SAMD21G18

dronecz, 2018-06-13:
Hello, 

I just started using Circuitpython in one project with RTC chip DS3231, but this chip is bit more expensive to use as I´m focusing this to be workshop for children, so I want to ask if is in CP support for RTC in SAMDG21G18. I saw commit #888 from @notro but we can not make it work.  

Any help or info about it is very welcome. At least I will not waste my time try to make it work. 

Many thanks.

ladyada, 2018-06-13:
nothing works as well as an off-chip RTC because it has its own battery... the DS3231 is expensive but hte PCF8523 is a lot less expensive! https://www.adafruit.com/product/3295

tannewt, 2018-06-14:
Hi @dronecz for tech support please post on the Adafruit forums: https://forums.adafruit.com/viewforum.php?f=60 Please post more info there about what you are trying to achieve, what code you have to do it and explain how its not working.

Issues are mainly for feature requests and bug fixes.

closed 2018-06-14 by tannewt