adafruit/circuitpython #931: Mu crashes REPL

BtreeComputingServcies, 2018-06-15:
Reference this...
https://forums.adafruit.com/viewtopic.php?f=50&t=132938&p=677479#p677479

Raspberry Pi Rx-> Cpx Tx
Raspberry Pi Tx -> Cpx Rx
And grounding pin on raspberry pi (39) to ground on cpx cause the crash immediately.

Ultimately, what I am trying to accomplish is a two way communcation between the pi and cpx using the serial pins.

I have no crashing issues if I do not use the ground, but then I am only able to send data from the Pi; the pi can not receive data from the cpx.

sommersoft, 2018-06-16:
@BtreeComputingServcies could you explain a little further what you mean by "crash"? It is a little hard to tell if this is a hardware issue, or a software issue.

Does the CPX reset completely (power cycle)? Or, does the program just stop running (with an Exception)?

If there is an exception, please post the Traceback.

BtreeComputingServcies, 2018-06-16:
I moved the ground to another pin on the raspberry pi and now it works correctly without any errors.

The message I saw in the REPL was that I was running in safe mode, but something serious had happened and to report it.  I should have copied the full content.  Feel free to close this issue.  I'll report back if it happens again.  Thank you for your consideration.