adafruit/circuitpython #954: PC freezes when upgrading from 2.3.1 to 3.0

tonym01, 2018-06-21:
I'm getting a problem trying to upgrade a metro m0 express from CircuitPython 2.3.1 to 3.0. I'm using a PC with windows 10 - I switch to the boot drive mode and copy the 3.0.0-rc.0.uf2 file from github into it. As the .uf2 file is copying to the board the PC locks-up (mouse and keyboard unresponsive). The board reboots but PC remains locked. If I reset the board the PC unlocks briefly then locks-up again. When I unplug the board's USB cable the PC unlocks, when I plug the cable back in again the PC locks-up.
I can copy the 2.3.1.uf2 back onto the boot drive and all works OK.

Windows PC details
HP Omen laptop 15-ce0xx intel core i7HQ @ 2.80GHz
Windows 10 Home 64-bit version 1803 OS build 17134.112

CircuitPython upgrade file name:
adafruit-circuitpython-metro_m0_express-3.0.0-rc.0.uf2

Metro M0 express details

METROBOOT folder INFO_UF2.TXT contents:
UF2 Bootloader v1.21.0 SFHR
Model: Metro M0
Board-ID: SAMD21G18A-Metro-v0

CIRCUITPY folder boot_out.txt contents:
Adafruit CircuitPython 2.3.1 on 2018-05-07; Adafruit Metro M0 Express with samd21g18

CIRCUITPY folder main.py
`import digitalio
import board
import time

led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT
while True:
    led.value = not led.value
    #print("Hi ")
    time.sleep(2.0)
`

tannewt, 2018-06-21:
Thank you for the report! We'll look soon!

tonym01, 2018-06-21:
No problem - let me know if you need any more information.

tonym01, 2018-06-22:
Thanks - I'll try the suggested ideas. I am away at the moment, so it will be tomorrow before I get back with my findings. I'll also try a different PC and see if I get the same behaviour.

tonym01, 2018-06-24:
Sorry for the delay in responding - please see my findings below;

I tried suggestion 1.  erasing circuitpy, but got the same behavior when copying the 3.0.0-rc.0.uf2.

I then tried this again with a windows 7 laptop and a freshly downloaded 3.0.0-rc.0.uf2 file:
The windows 7 PC did **not** freeze, but after copying the file to METROBOOT the CIRCUITPY drive would not appear (tried re-setting the board etc.), so there is still a problem. I doubt its related to drivers as this was a different PC, but I'll find a Macbook to try this on also.

Re. question 2: I don't have any third-party disk-indexing or backup programs installed: just the usual antivirus (Kaspersky).

Re. question 3: on the original windows 10 PC the METROBOOT drive does disappear, but CIRCUITPY does not re-appear.

I'll post an update after I've tried this on a Macbook.

dhalbert, 2018-06-24:
After you try on the Macbook, assuming it still has a problem, let's try updating the bootloader: you have a fairly old version. Drag this uf2 to METROBOOT, and wait for the flashing to finish and the red LED to start pulsing agin
https://github.com/adafruit/uf2-samdx1/releases/download/v2.0.0-adafruit.5/update-bootloader-metro_m0-v2.0.0-adafruit.5.uf2
Then you'll have to reload CircuitPython.

Is this the only CircuitPython-capable board you have, or do you have any others that seems to work fine?

tonym01, 2018-06-24:
I've not got my hands on a macbook yet, but I did update the bootloader as per your instructions. Unfortunately I'm still experiencing the same behavior.
I've got a GEMMA M0, so tried the 3.0.0-rc.0.uf2 for that board and got the same problem with the keyboard and mouse freezing on the windows 10 PC. On the GEMMA I noticed the buit-in dotstar LED  flashing blue about 8 or 9 times in succession - does this help?

tonym01, 2018-06-24:
The sequence of the flashing dotstar LEDs on the GEMMA M0 is GREEN(x1), CYAN(x1), BLUE(x9).
main.py has the following code;

`# Gemma IO demo
# Welcome to CircuitPython 2.0.0 :)

import board
import time
from digitalio import DigitalInOut, Direction, Pull
from analogio import AnalogIn, AnalogOut
from touchio import TouchIn
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
import adafruit_dotstar as dotstar

# One pixel connected internally!
dot = dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1, brightness=0.2)

# Built in red LED
led = DigitalInOut(board.D13)
led.direction = Direction.OUTPUT

# Analog output on A0
aout = AnalogOut(board.A0)

# Analog input on A1
analog1in = AnalogIn(board.A1)

# Capacitive touch on A2
touch2 = TouchIn(board.A2)

# Used if we do HID output, see below
kbd = Keyboard()

######################### HELPERS ##############################

# Helper to convert analog input to voltage
def getVoltage(pin):
    return (pin.value * 3.3) / 65536

# Helper to give us a nice color swirl
def wheel(pos):
    # Input a value 0 to 255 to get a color value.
    # The colours are a transition r - g - b - back to r.
    if (pos < 0):
        return [0, 0, 0]
    if (pos > 255):
        return [0, 0, 0]
    if (pos < 85):
        return [int(pos * 3), int(255 - (pos*3)), 0]
    elif (pos < 170):
        pos -= 85
        return [int(255 - pos*3), 0, int(pos*3)]
    else:
        pos -= 170
        return [0, int(pos*3), int(255 - pos*3)]

######################### MAIN LOOP ##############################

i = 0
while True:
  # spin internal LED around!
  dot[0] = wheel(i)
  dot.show()

  # set analog output to 0-3.3V (0-65535 in increments)
  aout.value = i * 256

  # Read analog voltage on A1
  print("A1: %0.2f" % getVoltage(analog1in))

  # use A2 as capacitive touch to turn on internal LED
  if touch2.value:
        print("A2 touched!")
        # optional! uncomment below & save to have it sent a keypress
        #kbd.press(Keycode.A)
        #kbd.release_all()
  led.value = touch2.value

  i = (i+1) % 256  # run from 0 to 255
`

tonym01, 2018-06-24:
sorry that didn't paste in very well - its the standard demo code that came pre-installed.

dhalbert, 2018-06-24:
(I edited the code paste: If you surround it in three backticks, it formats as code.)

Rename `main.py` to something else so that there is no main.py or code.py. And was there no main.py sometimes on the Metro M0 and it still locked up?

tonym01, 2018-06-25:
Hi, on the GEMMA M0 I renamed the main.py file. The PC still locks up after the 3.0.0-rc.0.uf2 file is copied to the GEMMABOOT drive, but now I get a green heartbeat on the dotstar LED, instead of blue blinking. 
I also tried switching off kaspersky, but that did not make a difference.
Still waiting to try this on a Macbook.

tonym01, 2018-06-25:
I've checked this on a Macbook now, and can report that the 3.0.0-rc.0.uf2 file works:  the CIRCUITPY drive appears and the Macbook operates normally. So it seems that this is a problem affecting windows, but not Macbooks.

dhalbert, 2018-06-25:
I'm guessing that it may very well be something you have installed on both your Windows machines that's getting confused. Perhaps go through everything that's installed and see what's third-party and common to both machines.

Another user had some monitoring software called "AIDA64" that caused trouble: https://forums.adafruit.com/viewtopic.php?f=60&t=135400

Also, as I mentioned, 3.0.0 newly supplies a gamepad HID device and a "consumer control" (media keys) device. It's  possible you have a weird game controller driver installed that's getting confused by the gamepad.

tonym01, 2018-06-26:
I restarted the windows 10 laptop in safe mode, and then plugged in the GEMMA M0 with circuitpy 3.0.0-rc.0 installed. The CIRCUITPY drive appeared and the laptop did not lock up, so it must be a driver-related issue. I'll see if I can isolate what is causing the conflict.

dhalbert, 2018-06-26:
Great, thanks! Another person is having similar issues here: https://forums.adafruit.com/viewtopic.php?f=63&t=134698&p=680535#p680503. They also have Kaspersky installed.

tonym01, 2018-07-01:
Hi. this does seem to be caused by Kaspersky - I re-tried disabling all services and startup programs and found that Kaspersky was still running - I had to manually exit the Kaspersky program to stop it from running. The Gemma M0 was then detected.

dhalbert, 2018-07-01:
Thanks for the testing! I did some brief research and I don't see this as a typically reported problem. If there are features of Kaspersky you can turn off, perhaps you can disable some but not all and get it to work. I'm not sure what would cause it to hang up the host computer. It may possibly have to do with the USB device switching identities under its nose, but that's just speculation.

tannewt, 2018-07-02:
Closing since this isn't a CircuitPython issue. Thanks for the good debugging!

closed 2018-07-02 by tannewt

tonym01, 2018-07-11:
I've logged this with Kaspersky support and they have acknowledged that this is a new bug. Apparently the developers are currently working on a fix. I'll keep this thread updated for others' information - when (or if) Kaspersky come back with a solution.

jay6621, 2019-01-19:
I am having exact same issue but on **Arduino Zero**. 
I flashed **adafruit-circuitpython-arduino_zero-3.1.2.bin** using **BOSSA 1.9.1** 
command line Log is as follows: 
```
C:\Program Files (x86)\BOSSA>bossac -e -w -v -R --offset=0x2000 adafruit-circuitpython-arduino_zero-3.1.2.bin
Erase flash

Done in 0.829 seconds
Write 188204 bytes to flash (2941 pages)
[==============================] 100% (2941/2941 pages)
Done in 2.302 seconds
Verify 188204 bytes of flash
[==============================] 100% (2941/2941 pages)
Verify successful
Done in 35.170 seconds
```
and Than windows locks/Freezes until I unplug Zero from it.

I am on Windows 10 Enterprise 64 bit , intel i5 , 8GB RAM.
I am using ESET antivirus (not kaspersky) tried disabling it but same issue.
I have not installed  AIDA64 nor Hard Disk Sentinel.
I was trying it because wanted to TASTE circuitpython.

Posted it here, if I can help in anyway.
Good work all of you.

dhalbert, 2019-01-19:
@jay6621 Better to ask this in https://forums.adafruit.com. Can you flash back to 2.3.1 using another computer? Try uninstalling ESET completely if you can, and check the list of installed programs for other suspicious stuff. Make sure there's no main.py or code.py, and do the `erase_filesystem()` stuff mentioned above. Also try on another computer.