adafruit/circuitpython #955: Add PWMOut.duty_cycle_fraction (between 0.0 and 1.0)

dhalbert, 2018-06-22:
@tannewt et al: what do you think about adding `PWMOut.duty_cycle_fraction`, which will allow a simpler way of setting a fractional duty cycle. it would be mplemented as `int(fraction * 65535)` plus range checking. So the end user doesn't need to remember 65535.

Similar functionality proposed in #85, but that was time-based rather than fraction. So maybe this has been sort of decided already.

dhalbert, 2018-06-22:
I would also add it to `adafruit_seesaw.pwmout.PWMOut`.

tannewt, 2018-06-25:
I don't really like having more than one way of doing something in the C api. Can you subclass it in Python and add an additional property instead? My goal is to use Python for the friendly API layer.

dhalbert, 2018-06-25:
I did add it to `adafruit_seesaw.pwmout.PWMOut`, as `.fraction` (less verbose). It's convenient for the crickit library. That's enough for now. I'll close.

dhalbert, 2018-07-19:
This came up again: suggested by @dastels in https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/247#discussion_r202186553. I think maybe we should reconsider, since pulseio.PWMOut is used so frequently as itself and not as a component. It's not that much code.

tannewt, 2018-10-09:
@dhalbert Do you still want this or can we close?

dhalbert, 2018-10-09:
Frankly, I would like it, to match seesaw's PWMOut. I see it more like different units, like fahrenheit and Celsius, or miles and kilometers, not as two different ways to do the same thing.

tannewt, 2018-10-09:
I've pushed to not have different units for the same reason. I'd rather write use-case specific wrappers that expose it as a value 0-1.0 similar to Servo and ContinuousServo.

dhalbert, 2020-11-01:
Closing for now, since we don't have consensus on this.

closed 2020-11-01 by dhalbert