adafruit/circuitpython #957: Minimizing for production

rucuriousyet, 2018-06-23:
Hi! I am working on an embedded project that would greatly benefit from using python for embedded sensors instead of C++ or Arduino. I have used the Micropython firmware on the ESP8266 platform and would love to use it on the SAMD21/Feather M0. My question is how one might build a port of circuitpython that removes all the extras such as filesystem mounting, REPL etc to create a minimal firmware image much better suited for devices in the field.

The goal would be to provide pre-compiled bytecode application with only the necessary dependencies freeing up more flash and ram for the application and potentially OTA updates over Lora (yes I know it would be very very slow).

dhalbert, 2018-06-24:
You are probably going to run out of RAM before flash becomes an issue. The best way to save RAM for program space is to use frozen modules. See https://learn.adafruit.com/building-circuitpython?view=all#adding-frozen-modules for how to freeze modules into the CircuitPython firmware.

In addition, consider using our M4 boards, including the Metro M4 Express, the Itsy Bitsy M4 Express, and arriving very soon, the Feather M4 Express. All have 256kB RAM (instead of 32kB on the M0), and 512kB flash (instead of 256kB).

tannewt, 2018-06-25:
We haven't really targeted a production environment with CircuitPython. However, we'd be happy to accept and pull requests for improvements you need.

If you are doing battery power then you'll likely need to improve the power usage as well. We haven't looked at it at all. Thanks!

arturo182, 2018-06-25:
I was actually thinking about this the other day, since polling and not interrupts are the CP way of working, it might be hard going to deep sleep, usually it's a external interrupt that wakes up the device, but in CP it looks like the device has to periodically wake up to see if anything happened and then decide if it should go back to sleep or act on the events.

dhalbert, 2018-06-26:
Closing. Reopen if you have more questions, or ask in #circuitpython on https://adafru.it/discord or in https://forums.adafruit.com

closed 2018-06-26 by dhalbert