adafruit/circuitpython #958: rtc does not work properly with M4

hotcarrier, 2018-06-25:
works fine with m0, but does not load hours and minutes for the m4

mrmcwethy, 2018-06-25:
@hotcarrier  Which clock board are you using?  Can you show a failing code example?

tannewt, 2018-06-26:
Its the internal RTC. This came up on the forum: https://forums.adafruit.com/viewtopic.php?f=60&t=137350

notro, 2018-06-26:
Indeed it's broken:
```
Adafruit CircuitPython 3.0.0-rc.0-25-gd7d132d5a on 2018-06-26; Adafruit Metro M4 Express with samd51j19
>>> import rtc, time
>>> clock=rtc.RTC()
>>> clock.datetime = time.struct_time((2018, 6, 25, 19, 59, 55, 0, 0, 0))
>>> print(clock.datetime)
struct_time(tm_year=2018, tm_mon=6, tm_mday=25, tm_hour=0, tm_min=0, tm_sec=0, tm_wday=0, tm_yday=176, tm_isdst=-1)
>>>
```

The problem is in asf4, replicating samd21 fixes it:
```diff
diff --git a/samd51/hpl/rtc/hpl_rtc.c b/samd51/hpl/rtc/hpl_rtc.c
index 84ef82a..37e48cc 100644
--- a/samd51/hpl/rtc/hpl_rtc.c
+++ b/samd51/hpl/rtc/hpl_rtc.c
@@ -139,7 +139,7 @@ int32_t _calendar_set_counter(struct calendar_dev *const dev, const uint32_t cou
 {
        ASSERT(dev && dev->hw);

-       hri_rtcmode0_write_COUNT_reg(dev->hw, counter);
+       hri_rtcmode0_write_COUNT_COUNT_bf(dev->hw, counter);

        return ERR_NONE;
 }
@@ -151,7 +151,7 @@ uint32_t _calendar_get_counter(struct calendar_dev *const dev)
 {
        ASSERT(dev && dev->hw);

-       return hri_rtcmode0_read_COUNT_reg(dev->hw);
+       return hri_rtcmode0_read_COUNT_COUNT_bf(dev->hw);
 }

 /**
```
COUNT_COUNT_bf looks at SYNCBUSY. common_hal_rtc_set_time() calls calendar_set_date() before calendar_set_time() so I guess set_time is lost when out of sync?

After fix:
```
>>> import rtc, time
>>> clock=rtc.RTC()
>>> clock.datetime = time.struct_time((2018, 6, 25, 19, 59, 55, 0, 0, 0))
>>> print(clock.datetime)
struct_time(tm_year=2018, tm_mon=6, tm_mday=25, tm_hour=19, tm_min=59, tm_sec=55, tm_wday=0, tm_yday=176, tm_isdst=-1)
>>>
```

I'll see if I can make a PR this weekend, unless someone else wants to fix it :-)

mrmcwethy, 2018-06-28:
Fixed in github adafruit/asf4.  Should be fix in circuitpython soon.

tannewt, 2018-07-02:
Thanks again @mrmcwethy and @hotcarrier for the report!

closed 2018-07-02 by tannewt