adafruit/circuitpython #963: Unable to mount SD card on nrf52832

jerryneedell, 2018-06-26:
I just built the feather nrf52832 board from master and tried to mount my SDCard as I have routinely on this board, but it failed with:

```
Adafruit CircuitPython 3.0.0-rc.0-27-g91427b0 on 2018-06-26; Bluefruit nRF52 Feather with NRF52832
>>> 
>>> 
>>> import sdmount_lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_lib.py", line 12, in <module>
OSError: [Errno 19] ENODEV
>>> 

```
Did something change? Or did something break?

jerryneedell, 2018-06-26:
Here is sdmount_lib.py 
```
jerryneedell@Ubuntu-Macmini:~/projects/feather52$ cat sdmount_lib.py 
import adafruit_sdcard
import busio
import digitalio
import board
import storage
import sys
# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.D27)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
sys.path.append("/sd/lib")
```

dhalbert, 2018-06-26:
is 3.x still ok?

jerryneedell, 2018-06-26:
Yes 3.x still works

arturo182, 2018-06-27:
Maybe I broke something with my SPI patch, I will try to debug the issue tomorrow.

arturo182, 2018-06-28:
Hmm, strange, I just tested with master and I was able to mount without any errors. You said 3.x still works, so I guess it's not your sdcard or the board you use for it.

I think the error comes from here https://github.com/adafruit/circuitpython/blob/master/shared-module/storage/__init__.c#L39 which would mean that the `vfs = ...` call fails and return `None`, could you verify by printing `vfs` before you call `mount`?

jerryneedell, 2018-06-28:
hmm ```
Adafruit CircuitPython 3.0.0-rc.0-27-g91427b0 on 2018-06-28; Bluefruit nRF52 Feather with NRF52832
>>> 
>>> 
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> import sys
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.D27)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> vfs
<VfsFat>
>>> storage.mount(vfs, "/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>> 

```

jerryneedell, 2018-06-28:
```
>>> dir(vfs)
['mkfs', 'open', 'ilistdir', 'mkdir', 'rmdir', 'chdir', 'getcwd', 'remove', 'rename', 'stat', 'statvfs', 'mount', 'umount', 'label']
>>> 
```

arturo182, 2018-06-28:
Hmm, I'll try also on the feather52832, the previous try was on a pca10056, maybe it makes a difference.

jerryneedell, 2018-06-28:
alao I am using SOFDEV 5.0.0  SD s132 in case it matters

jerryneedell, 2018-06-30:
Just tried mounting an SDcard on a PCA10056DK  -- works fine!
Still not working on the feather52840

jerryneedell, 2018-06-30:
Oe other test I did was to unplug the SD Card on the feather52840 -- it correctly reports No SD Card  -- so the SPI interface is communicating with with the baord. As Arturo pointed out the error is at the mount point.

PR #991 (2018-07-09): nrf: Fix SPI transfers with data larger than HW allows
Fixes #963
closed 2018-07-09 by tannewt