adafruit/circuitpython #980: How do you test your CircuitPython library?

notro, 2018-07-01:
I have started to work on an SMBUS slave library and I need a better way to do testing than the script I'm currently using.

I've spent some time playing with pytest and I can now run fixtures and tests on the board and exceptions are transferred and raised locally. 
The exception gets a custom traceback pointing to the correct source line.
Additionally I've done some simple assert rewriting, not as sophisticated as pytest which also avoids sideeffects.

So I think I have solutions for the hard parts in making a pytest plugin for running tests on the board itself through the repl.

Before I spend more time on this I'd be interested in hearing, how are people testing their libraries?

tannewt, 2018-07-02:
I usually just do manual testing. @cefn did more formal testing here: https://github.com/adafruit/Adafruit_Blinka/tree/master/test/src

notro, 2018-07-08:
Thanks @tannewt, that was a helpful example.
It helped me realise/decide that I have to upload the test modules/files to the board, instead of uploading and running the individual test functions over the repl.
(and I learned that there's a tool called ampy)

I have tried to avoid uploading files, because in my experience scripted upload over usb hasn't been that stable. Stability improved when I turned off automounting, but I believe it will be a hassle to support this reliably on linux/mac/windows.

So I'm now trying out uploading files through the repl with the flash writeable. It is slow.

I hope I can release an early version of the pytest plugin next weekend. But I suspect that things takes longer to do than hoped/expected, they usually do :-)

cefn, 2018-07-09:
For reference, the original commentary on the interactive testing (originally targeting very low resource ESP8266 Micropython boards) is at https://github.com/cefn/Adafruit_Micropython_Blinka/blob/master/examples/index.md

However, YMMV as the changes in this repo diverge from the Micropython-specific testing arrangements of https://github.com/cefn/Adafruit_Micropython_Blinka and the tests have never been run on RPi at all.

@notro I'm guessing you're running on much more beefy platforms is that right?

notro, 2018-07-09:
My main take away after skimming those tests, was that I should be able to support running tests on the blinka compatibility layer, at least make decisions that don't make it hard futher down the line.

If your word _platform_ refer to microcontroller/board, I'm doing this on a Feather M0 Express.

My pytest plugin will upload the test files to the board, but the test runner will run on the host not the board. The runner executes the test functions and fixtures (setup/teardwon) through the repl. Board exceptions are reraised on the host.

There are some upload scripts in the blinka repo. Have you experienced issues uploading files to the board? In my experience it's been somewhat flaky, but I'm new to this so I might be doing something wrong.
In an ideal world, I would have used the rsync protocol to sync the test files, avoiding upload and flash write if nothing has changed.

notro, 2018-07-13:
I've made the plugin public:
https://github.com/notro/pytest-circuitpython

It's my first attempt at Python packaging, but it seems to be installable.

Next step is to use it for an actual project :-)

kevinjwalters, 2018-08-23:
I started to write an emulation for a tiny library for an i2c device I wrote but I need to learn more about Python testing to refine this: https://forums.adafruit.com/viewtopic.php?f=60&t=137711&p=682274

closed 2021-08-12 by notro