adafruit/circuitpython #981: Add BME680 driver to Pypi

cemclellan, 2018-07-01:
Please add the BME680 CircuitPython driver to Pypi. Information below:

I am following the excellent Learning Guide CircuitPython on Linux and Raspberry Pi at https://learn.adafruit.com/circuitpython-on-raspberrypi-linux?view=all.

I have a BME680, not the BME280 and I would like to connect it to the Raspberry Pi using CircuitPython.

I installed the BME280 driver as per the instructions, hoping it might work, but I get the following error message:
RuntimeError: Failed to find BME280! Chip ID 0x61
[~/Python_Source/CircuitPython/BME280]

0x61 is the correct ID for the BME680 I have.

Apparently the BME280 driver is testing for 0x60.

ladyada, 2018-07-01:
OK try this!
https://pypi.org/project/adafruit-circuitpython-bme680/

cemclellan, 2018-07-01:
Wow! That was quick, thanks!!!

ladyada, 2018-07-01:
ok! closing, re-open if it doesnt work (we didnt test it, just deployed)

closed 2018-07-01 by ladyada