adafruit/circuitpython #996: Add board files for PCA10059 - the nRF52840 dongle

arturo182, 2018-07-06:
It's a nifty $10 board :)

ladyada, 2018-07-06:
@arturo182 hey! where'd you get that from?? :D

arturo182, 2018-07-06:
Straight from the source ;) https://twitter.com/arturo182/status/1014122869142491136

ladyada, 2018-07-08:
https://www.semiconductorstore.com/cart/pc/viewPrd.asp?idproduct=94964
fyi!

arturo182, 2018-07-11:
I came into possession of two more dongles, does anyone from Adafruit need some for testing? @ladyada @tannewt @hathach

jerryneedell, 2018-07-11:
@arturo182 I have some arriving in the next day or so (mouser just shipped ones that had been backordered) Be happy to test when the BSP is available.

arturo182, 2018-07-11:
Sweet! It's still on my TODO list, just a bit lower than other things ;) I think I'll wait for #1004 to be merged as this board doesn't have a easily accessible jlink interface.

dhalbert, 2018-07-11:
@arturo182 Thanks! A bunch of us ordered dongle samples from Symmetry; mine are out for delivery today.

hathach, 2018-07-11:
unfortunately, I didn't have one.

ladyada, 2018-07-11:
@hathach can you order from https://www.semiconductorstore.com/cart/pc/viewPrd.asp?idproduct=94964 ? this dongle isnt so important to support so its ok if it takes some time, i think just arturo likes it ;)

hathach, 2018-07-24:
@ladyada the shipping cost to me is like 3x the price of the board :( . Since it is not urgent, maybe the next time you send me the prototype of feather nrf52840, you could include this as well. Also please include the later revision of feather52832 ( need it for low power testing and coding tutorial https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/165 )

ladyada, 2018-07-24:
ok we are making the feather 52840 prototypes this week. when i have some ready we'll email together and get you a package with all the items :)

dhalbert, 2018-08-26:
I am including this in a revamp of pin definition files and also #1040.

dhalbert, 2018-09-17:
Fixed by #1158.

closed 2018-09-17 by dhalbert