adafruit/circuitpython #1001: float __eq__ handling differs between CircuitPython and CPython

notro, 2018-07-08:
I'm trying to implement [pytest.approx](https://github.com/pytest-dev/pytest/blob/master/src/_pytest/python_api.py#L295) for use with CircuitPython.
This is the example from the [documentation](https://docs.pytest.org/en/latest/reference.html#pytest-approx):
```
>>> from pytest import approx
>>> 0.1 + 0.2 == approx(0.3)
True
```

However it doesn't work, ```__eq__``` is not called when approx() is on the right side. It works for integers.

Test class:
```python
class Approx:
    def __init__(self, expected):
        self.expected = expected
    def __eq__(self, actual):
        print(self.expected, '__eq__', actual)
        return abs(self.expected - actual) <= 0.001
```

CircuitPython:
```
Adafruit CircuitPython 3.0.0-rc.0-25-gd7d132d5a-dirty on 2018-06-26; Adafruit Metro M4 Express with samd51j19
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class Approx:
===     def __init__(self, expected):
===         self.expected = expected
===     def __eq__(self, actual):
===         print(self.expected, '__eq__', actual)
===         return abs(self.expected - actual) <= 0.001
===

>>> 0.1 + 0.2 == Approx(0.3)
False

>>> Approx(0.3) == 0.1 + 0.2
0.3 __eq__ 0.3
True

>>> 3 == Approx(3)
3 __eq__ 3
True

>>>
```

CPython:
```
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> class Approx:
...     def __init__(self, expected):
...         self.expected = expected
...     def __eq__(self, actual):
...         print(self.expected, '__eq__', actual)
...         return abs(self.expected - actual) <= 0.001
...

>>> 0.1 + 0.2 == Approx(0.3)
0.3 __eq__ 0.30000000000000004
True

>>> Approx(0.3) == 0.1 + 0.2
0.3 __eq__ 0.30000000000000004
True

>>> 3 == Approx(3)
3 __eq__ 3
True

>>>
```

dhalbert, 2018-07-08:
Can you get away with having the `Approx()` only on the left-hand side?

notro, 2018-07-08:
I think I'd rather skip implementing ```pytest.approx``` since it wouldn't work as advertised. I'm making a pytest plugin for general use and someone is bound to put it on the right side with the possibility of it being an undetected false positive.
Maybe I could make an assert function that takes the operands as arguments.

Thank @dhalbert for a detailed answer.

closed 2018-07-08 by notro

notro, 2018-07-09:
> with the possibility of it being an undetected false positive.

What was I thinking, it will always fail.

Hmm, maybe I should implement it after all...