adafruit/circuitpython #998: Suggestion: Documentation for "Navigating the sources"

larryfast, 2018-07-08:
Hi Adafruit,
Great job on your documentation.  Very thorough. And the quality is really high! You've clearly got some great standards. However like any documentation it gets stale and the cost of maintenance just keeps getting bigger. So what do you do about stale documentation that's too expensive to keep up to date? Reference the source tree!

Two examples then a suggestion.

Example 1:  I went through 4 different wiki pages on your site and others trying to find the appropriate magic for "from machine import Pin; Pin(14, Pin.IN, Pin.PULL_UP)". Along the way there were pages referencing modules and syntax such as import board, digitalio and others.  This wasn't the first time I found myself in out-of-date-example-page hell.

Having found the right document I was still left wanting.  Eg.  Where are all the 8266 specific enumerations defined? How do I disable Pin.PULL_UP? Open source helps a lot but I'm struggling to find things like "Where is the import source for machine.py"

Example 2:  Tony DiCola created an AWESOME Video and Wiki page for the Feather I2C Motor shield. BUT it's old! The code in the video and on the page no longer work. 
https://learn.adafruit.com/micropython-hardware-pca9685-dc-motor-and-stepper-driver?view=all  
Since the add-on boards have their own Modules it was easier to find the code.  Then I found his "I refactored the whole module" commit in April.  This led me to the up to date Example. Tony updated the github repo Example but couldn't afford the time to update the Wiki page.

Do I have solutions? No. But here are some things that could improve the situation AND allow pages get older more gracefully.

IMO a couple things could improve the situation. First it would help -me- a lot if you created some documentation about the source tree itself. Seems appropriate given the novice audience you're often targeting. How are the sources organized? FAQs about where to find key pieces like hardware specifics. This will be especially helpful in the boundary layer between the CPython libs and pure python.

Then your other pages could include things like
   - When was this Example/Video created & tested?  BTW, great job testing everything!
   - What were the system versions and where were the sources at the time it was written?  
     This could be some boiler plate at the foot of each page.
   - Links to the sources and source FAQs.
     This could be a single link to a central "Finding stuff in code" page that's in the boilerplate.

Thanks for listening. Keep up the GREAT work!

ladyada, 2018-07-08:
hiya "from machine import Pin; Pin(14, Pin.IN, Pin.PULL_UP)" is micropython syntax, can you let us know wher eyou found this kind of text in our documenation/wiki because we don't support micropython
thx!

uoo, 2018-07-08:
I've been thinking along similar lines, after spinning my wheels for a while trying to figure out how to change the memory map (I wanted to use part of the FRAM in an MSP430FR as additional non-volatile RAM instead of ROM).  There are little pieces of code scattered all over that implements this, and finding them from a standing start is time-consuming.  I wonder what is the best place to put information like this.  I'm guessing just continue git style, and add some MarkDown pages to the source tree, where people who are looking at the source code are likely to find it?

tannewt, 2018-07-10:
We intentionally disable wikis on github for CircuitPython repos to ensure info isn't there. So please link to what you are using.

API reference docs live on ReadTheDocs here: https://circuitpython.readthedocs.io/

Tutorials are on Learn: https://learn.adafruit.com/category/micropython-slash-circuitpython

larryfast, 2018-07-13:
@ladyada  the Pin(14, Pin.IN) is on a micropython page, so no worries. Specifically: [https://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html](https://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html).
 I'm still getting my feet wet and using any data source that seems to work.

dhalbert, 2021-06-15:
Closing this as mostly obsolete. Please open a new issue if you have further ideas.

closed 2021-06-15 by dhalbert