adafruit/circuitpython #1003: nrf52832 does respond to control-C - sometimes

jerryneedell, 2018-07-09:
While testing the SPI and I2C interfaces to the STMPE610 Touch screen copntroller, I found that once I start my program running, I cannot use control-C to break out.  This occurs with pot the I2C and SPI versions of the driver.

Control-C does work under the REPL -- have not tried other sensors yet.

Revetring to 3.x -- control C works properly running the same code.

No exeption handling in the program.
```
import board
import digitalio
from adafruit_stmpe610 import Adafruit_STMPE610_I2C
i2c=busio.I2C(board.SCL,board.SDA)
st=Adafruit_STMPE610_I2C(i2c)
while True:
    if not st.buffer_empty:
        print(st.read_data())
```

arturo182, 2018-07-09:
Hmm, I wonder if my UART with nRFx rewrite can fix that.

tannewt, 2018-07-16:
@hathach Are you looking at this?

hathach, 2018-07-17:
@tannewt  ah no, I am resolving #1021 usb issue. Will look at this afterwards.

jerryneedell, 2018-07-19:
this appears to have been fixed. Ok to close?

jerryneedell, 2018-07-19:
Sorry I tested this on the pca10056 dk  board. Not on feather53832.

jerryneedell, 2018-07-19:
verified that this is still not working for feather 52832 -- sorry for the confusion.

arturo182, 2018-07-19:
Yes, since this is 52832, this must be in the UART driver not USB ;)

jerryneedell, 2018-11-15:
Just a reminder that this issue is still present on the feather_nrf52832. 
Control-c does not terminate an executing script.
The only way to stop a script is a hard reset.

tannewt, 2018-12-06:
@dhalbert Did you fix this?

jerryneedell, 2018-12-06:
I have been try to find the istbut have not been successful. If you have suggestions about where to look I’ll be happy to keep working on it.

dhalbert, 2018-12-07:
@tannewt I just fixed pin claiming for UART, but not UART reset. I think that's still to do.

jerryneedell, 2019-02-15:
since we have dropped nrf52832  -- should we close this?

dhalbert, 2019-02-15:
Since no one has complained about this in 52840, let's close.

closed 2019-02-15 by dhalbert