adafruit/circuitpython #1005: SPI baudrate of 100000 does not work reliably

jerryneedell, 2018-07-09:
Using the current master, I found that I was getting confusing data for the STPME610 touch screen driver on a feather52832. No errors reported, but data was clearly getting garbled.
The default baud rate for this device is 100000 (100K). I tried raising the baudrate to 1000000 (1M) and it works normally.  @jepler pointed out that this may be a know "feature" for the nrf52 and 100000 is just too slow.

The workaround is simple, I just wanted to alter folks and see how/if you wanted to deal with this

ladyada, 2018-07-09:
by chance does it work ok with M0 - like is it an stmpe thing or a nrf52 thing?
dont forget the stmpe uses mode 3 or 1 (not the 'standard' mode 0)

jerryneedell, 2018-07-09:
It woks on a feather m0 express. This driver has not been changed. I thought the stmpe610 used either mode 0 or 1 depending on its whim.

ladyada, 2018-07-09:
ok tagged as nrf52 - those peripherals could have a bug. thanks :)

arturo182, 2018-07-09:
While 100K is not a speed that the nRF supports, this code https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/busio/SPI.c#L39 should bump it to 125K and it should work as expected, but I will try to see if anything goes wrong when that happens.

@jerryneedell could you try testing with 250K and 500K ?

jerryneedell, 2018-07-09:
at 250K and 500K the performance is improved, but still gives bad readings. It works very well at 1M

dhalbert, 2018-07-09:
@jerryneedell do you have Saleae or similar to check whether the requested freq is what's actually coming out on the SPI clock pin?

jerryneedell, 2018-07-09:
@dhalbert yes - I'll try to make some measurements tonight or tomorrow.

jerryneedell, 2018-07-10:
@dhalbert I captured some samples with the Salea - the clocks are generally as commanded, but at 250 - I occasionally saw a longer cycle at 238Khz and at 500 and occasional longer cycle at 454KHz
I did not see any notable deviations at 1Mhz.

yk9, 2018-09-13:
im also using the nrf52832 with the STMPE610 . i can read the ID but im getting no data from the registers.
anyone had this problem?

yk9, 2018-09-13:
i think im using the last version , yes the baud is 1M
here is the example of the init:
void _SMTPE610_init(void)
{
    uint8_t d ;
    ts_hardware_init();
    writeRegister8(SYS_CTRL1, SYS_CTRL1_RESET); // 0x03 / 0x02
    for ( d=0; d<65; d++)
    {
    readRegister8(d);
    }   
    writeRegister8(SYS_CTRL2, 0x00);      // Enable TSC and ADC Blocks
    writeRegister8(TSC_CTRL, 0x04);      // X, Y mode
    writeRegister8(TSC_CTRL, 0x05);      // Enable Device, No tracking index, X, Y mode
    writeRegister8(INT_EN, 0x01);         // TOUCH_DETECT-Interrupt are enabled

    writeRegister8(INT_EN, INT_EN_TOUCHDET);
    writeRegister8(ADC_CTRL1, ADC_CTRL1_10BIT | (0x6 << 4)); // 96 clocks per conversion
    writeRegister8(ADC_CTRL2, ADC_CTRL2_6_5MHZ);
    writeRegister8(TSC_CFG, TSC_CFG_4SAMPLE | TSC_CFG_DELAY_1MS | TSC_CFG_SETTLE_5MS);
    writeRegister8(TSC_FRACTION_Z, 0x6);
    writeRegister8(FIFO_TH, 1);
    writeRegister8(FIFO_STA, FIFO_STA_RESET);
    writeRegister8(FIFO_STA, 0);    // unreset
    writeRegister8(TSC_I_DRIVE, TSC_I_DRIVE_50MA);
    writeRegister8(INT_STA, 0xFF); // reset all ints
    writeRegister8(INT_CTRL, INT_CTRL_POL_HIGH | INT_CTRL_ENABLE);

jerryneedell, 2018-09-13:
ah - you are using the Arduino IDE -- My experience has been with the CircuitPython Library. I have used it on an nrf52832 with CircuitPython.

yk9, 2018-09-13:
not arduino - IAR

jerryneedell, 2018-09-13:
Ah - sorry --- Still , the discussion above was with respect to the CircuitPython implementation.

jerryneedell, 2018-09-13:
I really am not sure what your code is doing and what you mean by 

>  i can read the ID but im getting no data from the registers.

does readregister8(d)  -- display the value?

yk9, 2018-09-13:
yes im getting some values . but its the initialization

jerryneedell, 2018-09-13:
Are you using the SPI or I2C interface?  The SPI can be tricky since it can sometime use either clock edge when it initializes.  the the cod in the Arduino Library detects this and switches modes:
see https://github.com/adafruit/Adafruit_STMPE610/blob/master/Adafruit_STMPE610.cpp 
I'm not sure if that is related to your problem or not.  - You may want to post your questions to that repository to see if anyone has any suggestions.

yk9, 2018-09-13:
using SPI. im controlling the SS so data is control. since im getting the correct data (ID and VER) i guess that the SPI is correct. am i wrong?

jerryneedell, 2018-09-13:
Agreed, If you can read the ID then it is not clear why you can't read other registers. I'm not sure what else to suggest.  
I suggest posting this in the Adafruit Forums https://forums.adafruit.com/viewforum.php?f=19

From what you have posted, it is still not completely clear to me what is not working. Can you post an example of the code that is not working?

tannewt, 2019-02-05:
@jerryneedell Is this still an issue?

jerryneedell, 2019-02-05:
no -- tested with partial_xenon at 100000 baud rate -- worked fine -- BTW the default is now 1Mbps
both work

closed 2019-02-05 by jerryneedell