adafruit/circuitpython #1007: USB serial

tannewt, 2018-07-10:
CircuitPython REPL accessible via native USB (nRF plug on DK).

dhalbert, 2018-08-27:
@hathach both @ladyada and i can't get to the REPL from Windows, LInux or Mac, when loading the latest master on pca10056. Is this supposed to work yet? Serial connects but doesn't respond.

hathach, 2018-08-27:
Hmm, weird, I don't have any issues connecting on linux, only a bit of delay for the first connection. Which is the previous known bug. Maybe it is time to troubleshoot it as well.

hathach, 2018-08-27:
analyzer show a huge delay for set baudrate request, after the serial is connected(). I guess the serial_connected() return true, trigger some function that didn't run the background task to handle the baudrate request !!!. Linux seems to be more patient, windows and mac seems to timeout and drop connection. Looking more closely at it now  

[image]

hathach, 2018-08-27:
@dhalbert  @ladyada  
More on troubleshoot. If we comment out these message function in main.c::run_code_py(), it will connect instantly. My guess is the below code try to write data to the cdc before the above set baudrate request is complete. it is still valid since the LINE STATE request  is complete indicating terminal is connected. 

Due to the nrf's usb easy DMA resitrction that **only 1 DMA transfer can be active at a time**, this seems to block the control response which is serve in the background task in `MICROPY_VM_HOOK_LOOP` until all data in cdc transferred is complete.  kaka the mcu have to run  fast to run into this type of issue. At least there is no mystery bug now, crossing finger to think of way to address this nicely.  

```
    while (true) {
        #ifdef MICROPY_VM_HOOK_LOOP
            MICROPY_VM_HOOK_LOOP
        #endif
        if (reload_requested) {
            return true;
        }

        if (serial_connected() && serial_bytes_available()) {
            // Skip REPL if reload was requested.
            return (serial_read() == CHAR_CTRL_D);
        }

        if (!serial_connected_before_animation && serial_connected()) {
//            if (serial_connected_at_start) {
//                serial_write("\n\n");
//            }
//
//            if (!serial_connected_at_start) {
//                if (autoreload_is_enabled()) {
//                    write_compressed(translate("Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.\n"));
//                } else {
//                    write_compressed(translate("Auto-reload is off.\n"));
//                }
//            }
//            // Output a user safe mode string if its set.
//            #ifdef BOARD_USER_SAFE_MODE
//            if (safe_mode == USER_SAFE_MODE) {
//                serial_write("\n");
//                write_compressed(translate("You requested starting safe mode by "));
//                serial_write(BOARD_USER_SAFE_MODE_ACTION);
//                serial_write("\n");
//                write_compressed(translate("To exit, please reset the board without "));
//                serial_write(BOARD_USER_SAFE_MODE_ACTION);
//                serial_write("\n");
//            } else
//            #endif
//            if (safe_mode != NO_SAFE_MODE) {
//                serial_write("\n");
//                write_compressed(translate("You are running in safe mode which means something really bad happened.\n"));
//                if (safe_mode == HARD_CRASH) {
//                    write_compressed(translate("Looks like our core CircuitPython code crashed hard. Whoops!\n"));
//                    write_compressed(translate("Please file an issue here with the contents of your CIRCUITPY drive:\n"));
//                    serial_write("https://github.com/adafruit/circuitpython/issues\n");
//                } else if (safe_mode == BROWNOUT) {
//                    write_compressed(translate("The microcontroller's power dipped. Please make sure your power supply provides\n"));
//                    write_compressed(translate("enough power for the whole circuit and press reset (after ejecting CIRCUITPY).\n"));
//                }
//            }
//            serial_write("\n");
//            write_compressed(translate("Press any key to enter the REPL. Use CTRL-D to reload."));
//        }
//        if (serial_connected_before_animation && !serial_connected()) {
//            serial_connected_at_start = false;
        }
        serial_connected_before_animation = serial_connected();

        tick_rgb_status_animation(&animation);
    }
```

hathach, 2018-08-28:
The issue is a bit complicated, but require only a few lines to fix as expected. The issue is EP0STATUS to send control status also require Easy DMA access. This seems to be missing from Nordic documents

https://github.com/hathach/tinyusb/compare/7b35cd0203bc409d7c1aefc075672103cb4a913e...583326e535454f16b06ebdb9cc06869602a5564c#diff-9b5241cf0c81e9eb639b442c58170472R220

commit 93100a75 (2018-08-28): Merge pull request #1146 from hathach/nrf52_usb_serial_bug

update tinyusb to fix #1007 serial issue
commit ba26ca9e (2018-08-28): update tinyusb to fix #1007 serial issue
closed 2018-08-28 by dhalbert