adafruit/circuitpython #1009: USB HID

tannewt, 2018-07-10:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_hid/__init__.html#module-usb_hid

dhalbert, 2018-07-17:
split this into Bluetooth and wired USB support?

tannewt, 2018-07-17:
This is USB only. A BLE HID issue would be good too.

hathach, 2018-07-20:
Descriptors to support are
- keyboard
- mouse
- consumer key ( for volume up, down, brightness +/-)

dhalbert, 2018-07-20:
We have code to generate the USB descriptors in https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/tools/gen_usb_descriptor.py (most of which should maybe be moved out of the port-specific directory). For HID, that code calls the library code in https://github.com/adafruit/usb_descriptor(which is a submodule further up in the tree), including pre-made HID descriptors in https://github.com/adafruit/usb_descriptor/blob/master/adafruit_usb_descriptor/hid.py. Besides the ones you listed, we also support a generic gamepad. There is latent support for a touchscreen but it does not work on all OS platforms so I turned it off.

hathach, 2018-07-20:
Ah thanks, I am also quite familiar with hid descriptor, so it is not really an issue. Ah, almost forgot the gamepad. I will add it as well

dhalbert, 2018-07-20:
I figured you were an expert on that :grinning:. Let me know if you think there's something that could be improved in the current descriptors.

hathach, 2018-07-20:
It is not true, I just copy thing here and there and put them together :)) . The whole hid descriptor is too complicated for anyone to digest :) . I will try out the descriptor generator as well. Above descriptors are all that I worked with so far.

dhalbert, 2018-07-20:
If you want, you can use the descriptors in https://github.com/adafruit/usb_descriptor/blob/master/adafruit_usb_descriptor/hid.py. They are all lumped in a single-endpoint composite HID device. I spent a while tuning them up and making sure they work on all platforms.

hathach, 2018-07-24:
@tannewt  no problem, I will make the hid report descriptor changeable dynamically (you will need to update the configuration descriptor for new length as well). However, to reduce amount of work, and get this done quickly, we will still implement it with fixed descriptor at first. When needed, we will change it with a separated PR.

hathach, 2018-07-24:
@tannewt just out of curiousity, we want to change hid report dynamically is to switch on/offf keyboard, mouse, gamepad etc right ? I ask since I always want to have a high level of api as much as possible for tinyusb. Those can be achieved at some level with predefined hid descriptor, though the full hid parser is too much for a device to support.

PS for configuration descriptor it is already dynamic supported by tinyusb, it will reconfigure endpoints based on parsing config desc when enumeration :)

hathach, 2018-07-24:
for ref
- https://github.com/DIGImend/usbhid-dump
- https://github.com/DIGImend/hidrd

hathach, 2018-07-25:
@tannewt I see, class dynamic enable/disable is already support by tinyusb. python only need to feed the correct config descriptor. So it  is not a problem at all.

For HID, to construct report in Python, I feel like we don't need to dynamically change the report descriptor. I could be wrong, let's me get something running first to see what it turns out to be.

dhalbert, 2018-07-25:
For HID, we mean adding or dropping devices from the multiple report descriptor: https://github.com/adafruit/usb_descriptor/blob/master/adafruit_usb_descriptor/hid.py#L170. E.g. turn off keyboard, or allow a user to add another device with its own Report ID. As you mention, this affects the length field in the composite descriptor. (This stuff was clearly designed by two different committees: it would be a lot easier if the length were in the header of the HID Report Descriptor :frowning_face:)

hathach, 2018-07-25:
@dhalbert  thanks, I already got the idea, the C level should only need to implement a generic HID where user need to supply report descriptor and report, which I will added to the stack.

FYI, In my stack, originally it attempts to support higher API with both keyboard & mouse API. It is useful to user that is not familiar with  HID and only want to use out of the box keyboard/mouse.
https://github.com/hathach/tinyusb/blob/develop/src/class/hid/hid_device.h#L65

hathach, 2018-07-25:
I also try to support boot keyboard at stack C-level as well, so it is a bit confusing with option and things. Trying to nail it now :D

dhalbert, 2018-07-25:
Yeah, I don't think we need boot keyboard support, but it could be moved out of the multi-device HID descriptor to its own endpoint. We have enough endpoints on the SAMD, at least.

dhalbert, 2018-07-25:
Note that you will not see this echoed in the REPL while it's running (if you just do `>>> k.send(Keycode.A)` you'll see it). But if you move the focus to another window you'll see 'a's being sent.
```
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
import time
k = Keyboard()
while True:
    k.send(Keycode.A)
    time.sleep(1.0)
```

hathach, 2018-07-25:
Superb, thanks. Sorry for silly question, I haven't used cp much :(

hathach, 2018-07-25:
:+1: :+1:

hathach, 2018-07-26:
@dhalbert  the readthedoc link should list `Adafruit_CircuitPython_HID` as dependency instead of the core repo right ?

hathach, 2018-07-26:
never mind, I think you mean the dependency of the HID library. Sorry, misunderstand there.

hathach, 2018-07-31:
@dhalbert @tannewt I would like to light up an LED when receiving SET REPORT request from host e.g CAPSLOCK. But couldn't find where to add it, could you please give me a hint ?

hathach, 2018-07-31:
keyboard and mouse works quite well, cleaning it up for full list of devices support

dhalbert, 2018-07-31:
> @dhalbert @tannewt I would like to light up an LED when receiving SET REPORT request from host e.g CAPSLOCK. But couldn't find where to add it, could you please give me a hint ?

We haven't provided a way to do that reverse-direction dataflow at all yet in the HID module, so it would be noticeable work and an API addition. Fine to skip it for now. You could open a long-term issue

hathach, 2018-07-31:
Ah, thanks. Since I don't know how to organize the cp API and my obj things. I will just turn on/off an LED0 for that, does it sound OK, if not I will leave it as stub.

dhalbert, 2018-07-31:
A stub is fine too, since we don't want to reserve an LED just for this. It's more likely LED state would be sent on to some peripheral, (e.g. a pin) for someone who's building a keyboard. Right now CircuitPython doesn't have an of handling callbacks which we might want for this case (since SET REPORT can arrive spontaneously).

hathach, 2018-07-31:
@dhalbert everything seems works, except for the gamepad, I got this error. I have no idea about this, am I missing something 

```
Adafruit CircuitPython 4.0.0-alpha-40-g2c85f42-dirty on 2018-07-31; PCA10056 with nRF52840                                                                                                                                                                                 
                                                                                                                                                                                                                                                                           
>>> from adafruit_hid.gamepad import Gamepad                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                           
>>> gp = Gamepad()                                                                                                                                                                                                                                                         
                                                                                                                                                                                                                                                                           
Traceback (most recent call last):                                                                                                                                                                                                                                         
                                                                                                                                                                                                                                                                           
  File "<stdin>", line 1, in <module>                                                                                                                                                                                                                                      
                                                                                                                                                                                                                                                                           
  File "adafruit_hid/gamepad.py", line 89, in __init__                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                           
  File "adafruit_hid/gamepad.py", line 86, in __init__                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                           
  File "adafruit_hid/gamepad.py", line 149, in reset_all                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                           
  File "adafruit_hid/gamepad.py", line 163, in _send                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                           
TypeError: 'bytearray' object does not support item assignment                                                                                                                                                                                                             
                                                                                                                                                                                                                                                                           
```

dhalbert, 2018-07-31:
>TypeError: 'bytearray' object does not support item assignment

In `mpconfigport.h`, try changing these from `0` to `1`:
```
#define MICROPY_PY_ARRAY_SLICE_ASSIGN            (0)
...
#define MICROPY_PY_BUILTINS_SLICE_ATTRS (0)
```
There are probably some other changes to the feature on/off flags that I should make match up with atmel-samd and esp8266. I'll do some diff'ing.

hathach, 2018-07-31:
Superb !! It is all working now, thanks. I will do some clean up and submit an PR for review

tannewt, 2018-07-31:
@hathach Whatever works. :-) Feel free to change it.

dhalbert, 2018-09-17:
Fixed by #1074.

closed 2018-09-17 by dhalbert