adafruit/circuitpython #1010: File transfer over BLE

tannewt, 2018-07-10:
Should feel like USB MSC but wireless including auto-reload

ladyada, 2018-07-10:
just a note, there doesnt seem to be a standard file transfer protocol - we'd have to hand-implement. not too hard but will need implementing per-platform
https://www.bluetooth.com/specifications/gatt

arturo182, 2018-07-10:
There is however a Object Transfer Service that could potentially be used for this https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.object_transfer.xml

tannewt, 2018-07-10:
@ladyada I realize that. Its just too handy to not have.

@arturo182 Thanks for the link. That looks promising.

arturo182, 2018-07-10:
I can have a look at this at some point, I think we should look into incorporating some parts of Nordic's BLE SDK as that would ease development and reduce the number of bugs we introduce when writing everything from scratch.

dhalbert, 2019-05-17:
I did some more looking at Bluetooth File transfer. There are some pre-existing specs There's OBEX, a general object exchange protocol which is used for IrDA, Bluetooth, USB, and other protocols. GOEP (Generalized Object Exchange Protocol) is a Bluetooth profile based on on OBEX. There's a File Transfer Profile (FTP).

https://en.wikipedia.org/wiki/OBject_EXchange
https://en.wikipedia.org/wiki/OBject_EXchange#Protocols
https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#File_Transfer_Profile_(FTP)

ladyada, 2019-05-20:
@hathach do you have any thoughts on which one of these would be most universal or best to implement?

ladyada, 2019-05-20:
https://www.howtogeek.com/208365/how-to-use-bluetooth-file-transfer-between-os-x-and-android-5.0-devices/

hathach, 2019-05-21:
So far I didn't know/work with a way to transfer file via ble yet. Haven't got any request for this in Arduino

alexwhittemore, 2020-12-17:
I might be inclined to work on this a bit if I can't easily satisfy my use case with a hackier system, so I'd love to collect some more thoughts about how this could/should work.

In my mind I've got two clear use cases: 1) something like image transfer, where a device with a camera needs to backhaul a small .jpg or something for processing on a gateway. 2) (my current need) in-the-field live firmware update. A gateway advertises some kind of FW update service where a device with a lower version can request the new `code.py`, copy it over, and restart. 

I'm working on a many-node sensor prototype. I'm not sure how long I'll stick with CP for iterating (eventually, code size will be a serious constraint), but if I make it to tens of units, I'll definitely need to put together something where I can push out a `code.py` update without manually plugging in each sensor for reprogramming. 

For that use case, it might even be fine to use nordic uart, write the text block out to disk, then do a rename and exit - no idea how that works in practice but a full file transfer service might be overkill. If it's not though, how do y'all envision such a service working? I know @tannewt mentioned in #3707 wanting to integrate this into the circuitpython core so it's available outside of the VM. One thing I'm very much NOT familiar with is the overall architecture and strategy of circuitpython implementation. Is there any under-the-hood documentation for would-be contributors? 

Alternately, writing it all in Python and packaging it up as a library seems more my speed. But the questions then become 1) how useful is that, 2) how practical is that (i.e. memory constraints), 3) what comparable BLE service implementations exist to use as a style guide?

tannewt, 2020-12-22:
> In my mind I've got two clear use cases: 1) something like image transfer, where a device with a camera needs to backhaul a small .jpg or something for processing on a gateway. 2) (my current need) [...]

I'm picturing that the device will have a BLE Server that allows a Client to read and write its files. The "hey I have a new image" would have to be done separately.

> For that use case, it might even be fine to use nordic uart, write the text block out to disk, then do a rename and exit - no idea how that works in practice but a full file transfer service might be overkill. If it's not though, how do y'all envision such a service working?

I'm picturing a Service with two characteristics. One is a simple number that is the version number. The second is a bidirectional service used to transfer the file info. It would be used similar to UART where it carries a custom protocol. (I don't want to overload UART directly though.) The client then supplies an action and a full path. Paths are always complete because you don't need to maintain current directory state.

If the client is reading it'll then send the number of bytes it can buffer. The server will return the total number of bytes, the number of bytes in response and then the bytes themselves. The client can then request more bytes by sending another number of bytes it can read or zero to stop reading. The server will return zeroes once it is at the end of the file.

Writing is similar except the server replies with it's free space and the client provides the length to be written.

We'll also need a way to list a directory. I think a json list is simplest but I'm open to ideas. Basically it'd work like file reading and writing but have a specific form for a directory.

> I know @tannewt mentioned in #3707 wanting to integrate this into the circuitpython core so it's available outside of the VM. One thing I'm very much NOT familiar with is the overall architecture [...]

It's not super detailed. I began sketching this out ages ago: https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/bluetooth.c

> Alternately, writing it all in Python and packaging it up as a library seems more my speed. But the questions then become 1) how useful is that, 2) how practical is that (i.e. memory constraints), 3) what comparable BLE service implementations exist to use as a style guide?

You could definitely prototype the service in pure Python. You'll be ok until a code.py update breaks the updater.

I'd probably model it after the [MIDI service](https://github.com/adafruit/Adafruit_CircuitPython_BLE_MIDI/blob/master/adafruit_ble_midi.py#L43) because it uses one bidirectional characteristic and a packet buffer for buffering.

alexwhittemore, 2020-12-22:
Tons of good food for thought, thanks!

tannewt, 2021-07-13:
This was added in #4918 and uses the protocol defined here: https://github.com/adafruit/Adafruit_CircuitPython_BLE_File_Transfer

closed 2021-07-13 by tannewt

alexwhittemore, 2021-07-13:
So rad. Thank you!

Geromino, 2025-04-19:
Hi @tannewt  , 
could help understand circuitpython behavior regard ftp service 

What’s happening in your setup:

You're running CircuitPython 9.2.7 on a Feather nRF52840 Express.

Your code.py uses BLE only for advertising, but no explicit BLE file transfer code is included.

The Glider app connects, and you can:

Create folders

Write files

After each file/folder operation (when running on battery, not USB), the device:

Performs an automatic soft reset

On reconnect (reopen app), the written data appears and is correct

Attached code.py 
import asyncio import time import _bleio import neopixel import os import board import busio import digitalio import gc from adafruit_ble import BLERadio from adafruit_ble.advertising.standard import ProvideServicesAdvertisement mac_address = 0 sw_version = "v1_2_9" def parse_uname_output(uname_output): return {'version': uname_output[3] if len(uname_output) > 3 else None} # uname_output-take only version uname_output = os.uname() fw_version = parse_uname_output(uname_output) def get_final_string(): global mac_address mac_address = str(_bleio.adapter.address).replace("<Address ", "").replace(">", "") # print(f"[sensera app reads]: {mac_address}") easyg_device_name = "EasyG" + "".join(mac_address.split(":")[-4:]) print("EasyG Name",easyg_device_name) return easyg_device_name colors = { "white" :(255, 255, 255), # white "off" :(0,0,0), "red" :(255,0,0), "green" :(0, 255, 0), "yellow" :(255, 255, 0), "orange" :(255,165,0), "blue" :(0, 0, 255) } pixels = neopixel.NeoPixel(board.NEOPIXEL,1,brightness=0.2) ble = BLERadio() ble.name = get_final_string() advertisement = ProvideServicesAdvertisement() easystatus = { "easystatus": { 1: "TBD", #Heart_rate_monitor 2: 3, #Vibration_pattern 3: 2, #Suction_intensity 4: 3, #Vibration_intensity 5: 0, #External_target_lubrication 6: 0, #Internal_target_lubrication 7: 25.0, #temperature_external_target 8: 25.3, #temperature_internal_target 9: 25, #External_heater_T 10: 25, #Internal_heater_T 11: 0, #Pressure_sensor_L 12: 0, #Pressure_sensor_R 13: mac_address,#MAC 14: False, #Sensera_Status 15: None, #battery_level 16: None #charge_status } } easystatus1 = { "easystatus1": { 17: fw_version, 18: sw_version, 19: None, #wifi_version() 20: "internal", #choosen_pump: "internal"/"external"/"both" 21: 0, #power_button_count (1-8) 22: 0, #drop_button_count (1/0) 23: 0, #touch_pin1.raw_value 24: 0, #touch_pin2.raw_value 25: 0, #gyro 26: 0, #Accelerometer 27: 0, #System_Current 28: 0, # touch1_average 29: 0, # touch2_average 30: 0 # current_avg_cap } } if ble.advertising: print("Auto-advertising detected. Stopping it...") ble.stop_advertising() def log_event(message): try: with open("/log.txt", "a") as f: timestamp = time.monotonic() f.write(f"[{timestamp:.1f}] {message}\n") except Exception as e: print(f"Failed to write to log: {e}") async def ble_handler(): toggle_debug = True while True: log_event("ble_handler: Top of loop") gc.collect() print("Free mem:", gc.mem_free(), "Allocated:", gc.mem_alloc()) if not ble.connected and not ble.advertising: ble.stop_advertising() print("stop Advertising...") ble.start_advertising(advertisement) print("start Advertising...") #await sensera_debug_log("start Advertising") if ble.connected: conn = ble.connections[0] print("Connected to central!", conn) while conn.connected: #device_cloud_commands = senserainfo.settings #senserainfo.easygstatus = easystatus #senserainfo.easygstatus1 = easystatus1 log_event("keep alive ble_handler device conected") print("keep alive ble_handler device conected") if toggle_debug : pixels[0] = colors["orange"] print("orange on") else : pixels[0] = colors["off"] print("orange off") toggle_debug = not toggle_debug await asyncio.sleep(0.5) # Avoid busy-waiting print("Connection lost!") #await sensera_debug_log("Connection lost") ble.stop_advertising() if toggle_debug : pixels[0] = colors["green"] print("green on") else : pixels[0] = colors["off"] print("orange off") toggle_debug = not toggle_debug log_event("keep alive ble_handler device not connected") print("keep alive ble_handler device not connected") await asyncio.sleep(2) async def main(): ble_task = asyncio.create_task(ble_handler()) tasks = [ble_task] try: await asyncio.gather(*tasks) except Exception as e: print("Top-level crash:", e) log_event(f"Top-level exception: {e}") while True: pixels[0] = colors["red"] time.sleep(0.5) pixels[0] = colors["off"] time.sleep(0.2) if __name__ == "__main__": log_event("System booted. Running main.") pixels[0] = colors["yellow"] time.sleep(0.5) pixels[0] = colors["off"] time.sleep(0.2) try: asyncio.run(main()) except Exception as e: log_event(f"asyncio.run crash: {e}") 

In boot.py add disable usb stoarge

My question is why device preform auto reload or sw reset even the copy file is not code.py

tannewt, 2025-04-21:
> My question is why device preform auto reload or sw reset even the copy file is not code.py

This matches USB behavior because it doesn't know what file was modified.