adafruit/circuitpython #1012: Use clang-format to enforce code conventions

arturo182, 2018-07-10:
As @tannewt mentioned in another issue, this would make sure that all code is uniformly formatted.
So let's have a discussion about how to implement this and which code style to use.

For the implementation we should run it as a part of the Travis CI for each PR for sure, but is there a way to run it also on users' computers before they commit. It can be done with pre-commit git hooks but those need to be setup manually by each user which is not great.

As for the code style, we can either use a predefined format or make our own, more info here: https://clang.llvm.org/docs/ClangFormatStyleOptions.html

Thirdly, the _Code Conventions_ link in https://github.com/adafruit/circuitpython/blob/master/CONTRIBUTING.md links to the micropython file, as it seems we don't have one in our repository, should we copy it locally, change the link and eventually modify the style to match our decisions?

And lastly, to what extent should the clang-format be applied, don't think we can expect all of the C code to be properly formatted, should it only be shared-bindings, shared-modules and the nrf and samd ports, or more?

dhalbert, 2018-07-10:
My major worry is making sure it's easy to continue to merge upstream micropython/micropython. It might be necessary to `clang-format` the upstream code before the merge; up to now our coding style has matched upstream and we haven't had to do something like that.

My major requirements are four-space indents, an opening curly brace does not stand alone, and always use curly braces in control structures. These are already what we do. For everything else I don't care very much, and don't actually think there needs to be an enforced style. For instance, how to indent arguments on multiple lines is not a hot button for me, nor line length, nor comment indentation or alignment. For `case` statements I use the standard style Emacs provides (which seems to match the current code), but I don't have a strong feeling about that.

dhalbert, 2018-07-10:
I went over the micropython conventions, and I'm completely happy with them (e.g. `//` not `/* */` comments)

arturo182, 2018-07-10:
I agree on priorities with @dhalbert , I think agreeing on how switch/case/break is aligned would be nice as well :)

tannewt, 2018-07-10:
I started experimenting with this here: https://github.com/tannewt/circuitpython/tree/clang-format

Merging should be easy as long as we stick with choices that can be fixed programmatically. (This is the goal anyway so we no longer need to worry about style.)

We'll need to pick a specific version of `clang-format` that is available on Travis.

dhalbert, 2020-07-21:
MicroPython has done this: https://github.com/micropython/micropython/issues/4223

tannewt, 2020-07-21:
I don't think that MicroPython is using clang-format. They are using astyle IIRC.

microDev1, 2021-03-21:
Closing... code formatting using _Uncrustify_ for `c` and _Black_ for `python` implemented in #4362.

closed 2021-03-21 by microDev1