adafruit/circuitpython #1016: digitalio example fails on Huzzah

larryfast, 2018-07-12:
When following the digitalio example on https://circuitpython.readthedocs.io/en/latest/shared-bindings/digitalio/__init__.html I get the following error:

>>> from board import *
>>> import digitalio
>>> 
>>> pin = digitalio(D14)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'D14' is not defined

I get the same error for any value of Dnn

Environment:
Clean install of CircuitPython 3.0.0 w. 3.0.0-beta Bundle 
Target: Huzzah
Install Method:  esptool.py & ampy

Note that this problem is also present in 2.3.1.  I reinstalled to verify the issue on a clean install of the latest code.

Cheers,
Larry Fast

jerryneedell, 2018-07-12:
The ESP8266 uses pin names GPIOxx rather than Dxx.
Try GPIO14

sommersoft, 2018-07-12:
Also, if you're ever curious about any module's available functions/objects, you use `dir(<module>)` to get a list.

For example, all of the pin names can be seen with `dir(board)`.

larryfast, 2018-07-12:
Thanks for dir().  I've been trying to track down python's "list internals of xyz" function for a couple weeks.

larryfast, 2018-07-12:
Since this is probably a documentation issue not a source issue, should I close it?

sommersoft, 2018-07-12:
Well, we could maybe still use the issue to improve the documentation on the ESP's different pin naming. In your experience so far, do you have a recommendation for a location(s) to document. Understand that some are easier to accomplish than others, but we should still strive to make it work.

larryfast, 2018-07-12:
As a reasonably green Python programmer I think I represent a lot of your audience. It would be VERY helpful to have a reference to dir() somewhere. Eg. a generic page about diagnosing python errors and especially some help on learning to interpret the C sources.

As for this particular issue, I think the digitalio page should say something about this board-to-board difference.

Thanks, Larry Fast

ladyada, 2019-08-23:
hiya, we no longer support the esp8266 as of 4.x - please check out micropython for esp8266 support! :)

closed 2019-08-23 by ladyada