adafruit/circuitpython #1017: inheriting from dict is broken in 3.0.0

larryfast, 2018-07-12:
The following class creation code works in 2.3.1 but breaks in 3.0.0:

class AttrDict(dict):
    def __init__(self, *args, **kwargs):
        super(AttrDict, self).__init__(*args, **kwargs)
        self.__dict__ = self

I loaded 2.3.1 to verify that it works on that version.  3.0.0 claims 'super' is not defined but I will get the exact 3.0.0 error message shortly.

dhalbert, 2018-07-12:
(@larryfast put code blocks inside triple backticks to format nicely. I edited the OP.)

caternuson, 2018-07-12:
(@dhalbert can also add language specifier for syntax highlighting. I edited the edited OP.)  :smiley:

larryfast, 2018-07-12:
Thanks for cleaning up my post. I updated it with the actual 3.0.0 error message.

dhalbert, 2018-07-12:
OK, this broke because we turned on `MICROPY_CAN_OVERRIDE_BUILTINS` in 01443a9619e7cbedd1697418855243cd95e0e674. However, note that it still doesn't really work right even before that change. If you call `AttrDict([])` you'll get an error, though `dict([])` works fine.

In general subclassing from builtins does not work well in MicroPython/CircuitPython. This is a deliberate choice to save space and time.

I'm not sure how you're class is used. It looks like you're trying to override the normal attribute dictionary, but I'm not sure that will work in any case, due to MicroPython idiosyncrasies.

larryfast, 2018-07-12:
Thanks for the update. My main goal is to conveniently declare system structure with the more readable `dict.foo.bar.baz` vs `dict['foo']['bar']['baz']`. I use this extensively in not-so-real-time world. 

I assume the change to `MICROPY_CAN_OVERRIDE_BUILTINS` isn't going to get undone. Is the slightly-more-broken-than-before behaviour of dict subclasses likely to get fixed or go onto the ...oh-well... pile?  Not looking for anything solid, just a vague sense of probability.

Cheers,
Larry Fast

dhalbert, 2018-07-13:
It looks like you're using https://github.com/bcj/AttrDict.

Different ports of MicroPython and CircuitPython do or do not turn on `MICROPY_CAN_OVERRIDE_BUILTINS`. We did choose to turn it on. I think it's an unfortunate accident that it happens to break `dict`-subclassing, As I mentioned, there are already limitations on subclassing of builtins (search in the micropython issues list for `dict`, `builtin` and/or `subclass` to get some idea of other issues). We don't tend to work on core language semantic issues: we leave that to MicroPython, and concentrate on other things. Neither we nor MicroPython are trying to make a completely faithful copy of CPython, though we want to get as close as practical for ordinary use. (For instance, that's why we have renamed the `u*` modules from MicroPython to match the CPython names, and tried to make sure they are the same or a proper subset of the CPython functionality.)

That doesn't sound too promising from your point of view. However, looking at `AttrDict`, I think you could create a library with more or less the same behavior that doesn't actually subclass from `dict`, with duck-typing. You'd need to override `__getattr__` and  (`__setattr__` if you want mutability) to get/set the values from a separate dictionary, but I think it would not be too hard.

larryfast, 2018-07-13:
Thanks again and perfectly reasonable that this issue is out of scope for your port. As you said there are other ways to skin this cat.

closed 2018-10-03 by dhalbert