adafruit/circuitpython #1018: CircuitPython support for the SAMD21J18

ExperimentalZeros, 2018-07-12:
I would like to respectfully request support for the SAMD21J18 chipset if possible. 

   If such task becomes to much demand to your group/organization, Would it be possible to indicate where to begin and what files will require modification to get me started?

  My real absolute goal would be to be able to use JTAG to program this chipset with CircuitPython!..... in which I will not pursue if by any means breaches your open source policy.

Thank you kindly!

EZ

dhalbert, 2018-07-12:
It should not be that hard. In `ports/atmel-samd/boards`, make a new directory corresponding to your board. Change everything as appropriate: pin mappings, `SAMD21G18A` -> `SAMD21J18A`, etc. If there are extra timers, SERCOMs, etc. that are in the J version, but not G, you may need to change some limits somewhere.

See `boards/ugame10` as an example of a third-party board we've included in our source tree. If you succeed and your board design is available to others, we'd be quite amenable to incorporating it into the sources.

ExperimentalZeros, 2018-07-12:
Great!

   Thanks for your quick response, and in regards to the board being available to others: absolutely, that is a goal as well!

   I will start with the information you just provided.

   Thanks again.

EZ

dhalbert, 2018-07-12:
Check out these guides for build and github info:
https://learn.adafruit.com/building-circuitpython
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

tannewt, 2018-07-13:
Why leave this open?

dhalbert, 2018-07-13:
No strong reason. I thought I would close after a day or two. @ExperimentalZeros we can reopen if you come back with a pull request.

closed 2018-07-13 by dhalbert

ExperimentalZeros, 2018-07-13:
Gents,

   Let’s do that as I am in the process of following the directions revised by Mr. Halbert.

Please close until then if appropriate!

Thanks