adafruit/circuitpython #1020: busio.SPI() and bitbangio.SPI() are not drop-in replacements for each other

tdstranger, 2018-07-13:
Most of the associated methods are identical except for three:

readinto()
busio has these args: readinto(buffer, *, start=0, end=len(buffer), write_value=0)
bitbangio only has:     readinto(buf)

write() has the same problem.
busio:        write(buffer, *, start=0, end=len(buffer))
bitbangio:  write(buf)

Lastly, busio has frequency() but bitbangio doesn't at all.

The learn.adafruit.com documentation does allude to busio and bitbangio being replacements but doesn't mention any SPI issues, if I remember correctly.

In my instance this means that the library for sdcard use is broken for Huzzah esp8266.

Thank you,
Stranger

tannewt, 2018-10-09:
@caternuson are they the same now?

caternuson, 2018-10-09:
Don't think so. But am I looking in the right place?
write:
https://github.com/adafruit/circuitpython/blob/master/shared-bindings/bitbangio/SPI.c#L198
readinto:
https://github.com/adafruit/circuitpython/blob/master/shared-bindings/bitbangio/SPI.c#L223
no entry for frequency:
https://github.com/adafruit/circuitpython/blob/master/shared-bindings/bitbangio/SPI.c#L313

tannewt, 2018-10-09:
Yup, that's the right spot!

jepler, 2020-07-29:
This prevents bitbangio.SPI from being used with adafruit_sdcard.

jepler, 2021-11-30:
In https://github.com/adafruit/circuitpython/pull/4497 we added support for `write_value` .  At that time, adafruit_sdcard.SDCard + bitbangio.SPI were tested and able to mount an SD card.  (sdcardio, the internal implementation, only works with busio.SPI).  start/end are supported in `readinto` and `write_readinto` though still not in `write` for some reason.

So, I think adding start/end to `bitbangio.SPI.write` is the only thing missing.

PR #5641 (2021-12-01): bitbangio.SPI: Handle kwargs like busio.SPI
Closes: #1020
closed 2021-12-01 by tannewt