adafruit/circuitpython #1021: nrf pca10056 - control C will not break out of loop

jerryneedell, 2018-07-13:
After loading current master to pca10056 board (via MSC! ) control-C does not break out of a While True: loop

```
>>> while True:
...     pass
...     
...     
... 

```
cannot break out with control- c

Same result with a while loop in a script:
```
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.P0_13)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

Blink works fine, but can't break out

arturo182, 2018-07-13:
Any hunches @hathach before I dive in too deep? ;)

hathach, 2018-07-13:
quick check on linux with minicom does seems to work.  @jerryneedell which OS and terminal you are using

```
Adafruit CircuitPython 3.0.0-rc.0-148-gbfe14ff-dirty on 2018-07-13; PCA10056 with nRF52840

>>> 

>>> while True:

...     pass

...     

>>> 1+1
```

arturo182, 2018-07-13:
I'm also using Linux (Mint) and minicom and it doesn't work for me either :/

jerryneedell, 2018-07-13:
I am using screen in Linux Ubuntu 16.04

hathach, 2018-07-13:
I am on ubuntu 16.04 as well. Let's me test with screen

jerryneedell, 2018-07-13:
I tried minicom under Ubuntu 16.04 and it did not work for me. :-(

hathach, 2018-07-13:
Hmm, screen works with me if I just typed the above while True: then press Ctrl +C.

Though if I go into paste mode and execute, I couldn't Ctrl+C to terminate it anymore. Is it standard or bug as well
1. Ctrl +E to go into paste mode, 
2. paste your blinky loop. 
3. Then Ctrl + D to run. 

PS: I also find out a bug with paste mode, where some character is missing as well. Working on the fix for that

jerryneedell, 2018-07-13:
@hathach did you enter enough "returns" to get into the While loop? I thought 3 were needed after the pass

jerryneedell, 2018-07-13:
or even 4

hathach, 2018-07-13:
@jerryneedell  ah I see, after the 4th CR, it is hanged there. Sorry, I am not familiar with cp :( . Let's me check, may be some loop() is runing ,and the background task that handle usb transfer is not executed while the loop is running.

jerryneedell, 2018-07-13:
Whew! Glad we are all seeing the same thing -- thanks!

sommersoft, 2018-07-13:
@hathach,
> PS: I also find out a bug with paste mode, where some character is missing as well. Working on the fix for that

We had a similar problem in SAMD land. Here was the fix:
https://github.com/adafruit/circuitpython/pull/698

Mention it as a possible place to look.

hathach, 2018-07-13:
Update: got it, the usb cdc doesn't scan and call mp_keyboard_interrupt() by now. Will try to find a way to add it in.

hathach, 2018-07-13:
@sommersoft  ah thanks, that will be very helpful :D

commit 3525d65a (2018-07-18): should fix #1021

- update tinyusb for wanted char
- move usb code into usb.c
closed 2018-07-18 by arturo182