adafruit/circuitpython #1025: ReadTheDocs structure issue.

three-bits, 2018-07-14:
This is a minor readability issue... it would be more clear in the Core Module section of the api ( http://circuitpython.readthedocs.io/en/latest/shared-bindings/index.html# ) if the API list(Modules) came before the support matrix(Support Matrix).

tannewt, 2018-07-16:
Good idea! Want to try changing it? Its here: https://github.com/adafruit/circuitpython/blob/master/shared-bindings/index.rst

GitHub guide is here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

three-bits, 2018-07-16:
Yes, I'll figure out how to fix this and do a pull request. I should be a good way to learn CircuitPython workflow.

I didn't see how to assign this to my self in git hub.

sommersoft, 2018-09-17:
Fixed by #1166. Thanks @dfarning!

closed 2018-09-17 by sommersoft